john30/ebusd-esp32 #2: manual read command not working

krolli53, 2023-06-12:
Hi,

i am using the ebus adapter V5 via wifi. Ebusd is installed on an ubuntu 20.04.6
Ebusd has a working connection and I get a list of all devices with command 'ebusctl find':

e.g.:
`broadcast load = no data stored
broadcast outsidetemp = 21.438
broadcast signoflife = no data stored
broadcast vdatetime = 08:55:09;12.06.2023
general valuerange = no data stored
memory eeprom = no data stored
memory ram = no data stored
scan id = no data stored
scan.04  = no data stored
scan.05  = Vaillant;VR921;2801;5703
scan.05 id = no data stored
scan.08  = Vaillant;BAI00;0807;7603
scan.08 id = 21;22;43;0010022008;0001;009276;N7
scan.15  = Vaillant;70000;0614;6903
scan.15 id = 21;22;37;0020266797;0082;057078;N4
scan.26  = Vaillant;VR_71;0100;5904
scan.26 id = 21;22;40;0020184846;0082;023631;N1
vr_71 currenterror = -;-;-;-;-
vr_71 errorhistory = no data stored
vr_71 SensorData1 = 23.44;-;-;-;59.06;49.44;52.00;54 00
vr_71 SensorData2 = 25.12;-;-;-;-895.81;0.00;80 54 04`

now I try to do 'sudo ebusctl read vr_71 SensorData1'

and it returns 
'ERR: element not found'

what am I doing wrong here?

krolli53, 2023-06-12:
update: after playing around with some parameters. It works only if I add the `-c` parameter. Is that normal?

`florian@odroid:~$ ebusctl read -c vr_71 SensorData1
23.44;-;-;-;58.44;51.00;62.06;54 00`

rom-burner, 2023-06-13:
yes, you are correct.  you are specifying the SensorData1 reading on the vr_71 circuit.  you might want to add the -f option as well.  the problem is the 'first steps' wiki is incorrect and misses the -c option, so any noob like me or you diligently following it gets caught for hours in this trap with unhlepful errors trying to figure it out!  unfortunately i cannot find the repo for the adapter website to raise a pull request to correct it....

john30, 2023-06-15:
if you'd look at the [wiki](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#read) or the help of the read command, then you'd see that indeed specifying the circuit name requires a "-c" in front

closed 2023-06-15 by john30

krolli53, 2023-06-15:
Sorry @john30 that I have bothered you with this stupid issue. May I ask you then to update the first steps section for the ebus adapter V5? There you say that if you do `ebusctl list` command and the response is for e.g. `hwc Mode = no data stored` you can just do `ebusctl read hwc Mode`. So I just followed the first steps and made that mistake. Sorry again.

john30, 2023-06-16:
thanks for spotting that, will correct it in the docs.
any many thanks for reading the docs! you seem to be a rare candidate :-)