john30/ebusd-esp32 #3: my_client_socket_send: send: errno 11 is a blocker to use the adapter?

JosemyAB, 2023-06-21:
Hello!
I just installed the new adapter v5 to my Saunier Duval Genia Set 8 and it works well connected via TCP with `ens` protocol to a raspberry pi zero which is running the version 22.4 of the ebusd program.

The only "weird" thing is that I get the error `my_client_socket_send: send: errno 11` in the adapater's log every one minute (more or less)
[image]
According with this other message https://github.com/john30/ebusd/issues/921 apparently it could be that the WIFI signal is not good enough or the "queue is full"

Well, the wifi signal is stable in -74dBm
[image]
Do you consider that is a poor signal?

On the other hand, is there any way to know if the "queue is full"? just to discard things.

And finally, could the error be dangerous for the machine? Should I avoid use it until I resolver the problem?

Thanks in advance

john30, 2023-07-01:
50% isn't really very good and you can expect signal outage. my advice is always to reach at least 75%

JosemyAB, 2023-08-03:
Hello again!
I have changed the position of the v5 adapter to increase the signal and now I usually have more than 75% of signal strengh
[image]

But I still having the same error:
[image]

I'm using the latest version of the adaptor's firmware and also the version 23.2 of the ebusd deamon running in a Docker container.
On the other hand, I'm using a 5v - 2A - 10W transformer as power source for the adaptor.

Can I do anything else to avoid the error?
Could it be dangerous for the machine?

Thanks in advance

john30, 2023-08-06:
if you see this message, then the wifi won't let you transmit the data from eBUS in the desired low latency of under 10ms. most probably this is due to some other traffic on the net. currently there is no means to retransmit later as the timing is ruined at that point already.

this will be addressed once full message transfer is implemented in ebusd

JosemyAB, 2023-08-07:
OK, so I will wait until new versions of ebusd to test it again.
Thanks for your response and for your effort to maintain this project, it is very useful!

john30, 2023-09-17:
should be fixed by now. could you check with the [new firmware 20230917](https://adapter.ebusd.eu/v5/ChangeLog#20230917) just released?

JosemyAB, 2023-09-26:
Thanks for the update! I will test it ASAP. Hopefully this weekend

john30, 2023-12-23:
didn't see this again with newer versions, so closing for now

closed 2023-12-23 by john30