john30/ebusd-esp32 #4: Umstellung auf USB ergibt "Bootloop"

Danit2, 2023-06-30:
Hallo John

Ich habe vor einer Woche meinen Adapter v5 erhalten.
Vielen Dank dafür. Ist ein gut gelungener Adapter.

Um ihn zu testen und erste Erfahrungen zu sammeln habe ich ihn per WLAN mit dem eBUSd verbunden.
Dies lief alles gut und ich konnte den Adapter ohne weitere Probleme konfigurieren und einrichten.

Nun habe ich die Zeit gefunden um mir ein BUS Kabel zu meinem Home Assistant zu legen.

Als ich mein Adapter aber nun auf USB Seriell umstellen wollte endete es in einem Bootloop.
Ich habe beide Firmware ausprobiert. Mit WLAN funktioniert alles ohne Problem und mit USB funktioniert nichts mehr. 
Ich muss anschliessend immer die Firmware neu flashen um überhaupt wider auf den Adapter zuzugreifen. 

Was kann ich da machen? Was ist hier das Problem? Ist der Adapter defekt?

Hier noch der Log während dem Bootloop:
[esp-web-tools-logs.txt](https://github.com/john30/ebusd-esp32/files/11919673/esp-web-tools-logs.txt)

Vielen Dank für die Unterstützung.

john30, 2023-07-01:
just tried it and can't reproduce. are you sure that there is no other process than ebusd using the virtual serial port?

Danit2, 2023-07-02:
Ok, I have now played around a bit.
The problem only occurs when the WIFI station is active and I switch the adapter to usb.
But the problem is also there when the adapter is only powered by a usb power supply.

If only the accsesspoint is active it works.

I have only one eBUSd this runs as an addon on homeassistant on a rpi4.

But now I have the problem that I always get the log message "scan config 08: ERR: read timeout" in the eBUSd.
What can I do? After about 30min the scan goes and the values are displayed. Via WIFI, the scans were always successful the first time.

The adapter with WIFI station and usb does not work. There is a bug in there somewhere.

john30, 2023-07-02:
not sure I got that right: why would you set it to usb and power it via usb? that wouldn't work anyway. or did you power via usb and use it via wifi?

again, i tried reproducing, flashed from scratch, connected to wifi, configured to usb, rebooted it and started ebusd using the usb serial. everything looks good and ebusd runs as expected using the device.
maybe you could note exactly the steps that you did.

whats your wifi strength btw? maybe thats not good enough (should be at 75% at least). bad wifi strength also may lead to protocol errors as it immediately impacts latency

Danit2, 2023-07-02:
Thank you for your answer.
I have now tried more things and could narrow down the problem a little.

The problem occurs when the adapter is set to usb and it is connected to my WIFI. As soon as I want to access the ip address, the adapter restarts.
When it is running again and no device is accessing the website the adapter is running.
As soon as a device wants to connect it crashes.

I think the problem is somewhere in my network.
Strange is that the problem only exists when the adapter is set to usb and not with WIFI connection.

Currently I have only the accesspoint active and the connection via usb works fine.

Unfortunately, I always get the following errors over the usb connection:

2023-07-02 14:05:29.903 [bus error] device status: buffer overflow
2023-07-02 14:05:39.165 [main error] scan config 08: ERR: read timeout
2023-07-02 14:05:41.252 [main error] scan config 15: ERR: read timeout
2023-07-02 14:05:43.372 [main error] scan config 76: ERR: read timeout

What does this mean? Why do these occur when connecting via usb. I did not have this when connecting via WIFI.

Thank you for your support.

john30, 2023-08-06:
so the ebusd connection is through usb, did I get that right?

Danit2, 2023-08-08:
Yes. It is over usb. 
But the problem with the boot loop is gone. 
My internet porovier send me a new router and now everything works fine. 

But about the errors. 
Can you make a site where you can explain all the errors that can happen on the bus? 
Or do you have this somewere?

john30, 2023-08-15:
what do you mean with "can happen on the bus"? if you're referring to errors that ebusd reports, check the ebusd repo and the [result.cpp](https://github.com/john30/ebusd/blob/master/src/lib/ebus/result.cpp) file

closed 2023-09-16 by john30