john30/ebusd-esp32 #5: Connecting to Wi-Fi network fails

brz, 2023-07-14:
I received my v5 in the mail yesterday but I am unable to connect it to my Wi-Fi network.
I installed firmware version 2023-06-18 and after select the Wi-Fi network and entering the password, I get the following screen:
[image]

Afterwards the connection times out:
[image]

I am 100% certain that the password is correct. My Wi-Fi network supports 2.4GHz and 5Ghz but the 5Ghz network is not stable. It support Wi-Fi N (not ac or ax). I run OpenWRT on my router.

Since this didn't work I tried connecting to the EBUS AP from my smartphone,
I can connect to the AP but when going to the web interface with http://192.168.4.1 the page keeps loading infinitely. I tried with Firefox and Chrome on my Android phone.

farfade, 2023-07-19:
Hello @john30,

I've got the same problem here.

Can you be more explicit on what security protocols are supported by the firmware for Wifi connexion ?
WPA2 ? WPA ? WEP ?

Note that WEP and WPA are weak and do not provide security.

Thanks for reading.

john30, 2023-08-06:
it supports what esp-idf does at the development branch of the build date, so there is no limitation on using 2.4 or 5 GHz e.g.

if you need a fix IP, you can connect to the easi AP "EBUS" and adjust the IP settings for station mode accordingly (IMRPOV does not support setting a fix IP).

john30, 2023-12-23:
fix ip for wifi was added some releases ago, failed connect logging should be there already. so please check again and if it still fails, then post the output of the log command via serial repl

john30, 2024-04-14:
closed for inactivity

closed 2024-04-14 by john30