john30/ebusd-esp32 #6: my_client_socket_send: send: errno 11

xpavli44, 2023-07-15:
Hey @john30 ,

I managed to connect your ebusd adapter 5 wifi adapter to home-assistant ebusd add-on. It seems to work at least partially. I am getting some values sent via MQTT successfully.

On the adapter side I see following error in logs
`my_client_socket_send: send: errno 11` 

I have no idea what that means and google does not help. Can you please tell me what that error means? 

[image]

Build: 20230618
[ebusd device string](https://github.com/john30/ebusd/wiki/2.-Run#device-options): ens:192.168.1.232:9999
WiFi station: ###redacted###, 192.168.1.232, 44% (-78dBm)
WiFi AP: inactive
Chip ID: dc5475ba43a0, [ESP32-C3, rev 4](http://192.168.1.232/#/status?verbose)
Hostname: ebus-ba43a0
Up time: 943930
Free heap: 117200
ebusd connected: yes
eBUS signal: acquired

I can see some errors on ebusd side as well but I am not sure if that is related or not

`2023-07-15 12:57:13.099 [bus error] poll bai PrEnergyCountHc2 failed: ERR: wrong symbol received
2023-07-15 12:57:15.707 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2023-07-15 12:57:19.260 [update notice] sent poll-read bai PrEnergyCountHc3 QQ=31: 0
2023-07-15 12:57:20.689 [update notice] received read bai Status01 QQ=10: 50.0;48.0;-;37.0;47.0;off
2023-07-15 12:57:22.660 [update notice] received update-read broadcast vdatetime QQ=10: 13:01:46;15.07.2023
2023-07-15 12:57:25.290 [update notice] sent poll-read bai PrEnergyCountHwc1 QQ=31: 1993671
2023-07-15 12:57:25.720 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2023-07-15 12:57:30.727 [update notice] received read bai Status01 QQ=10: 50.0;48.0;-;37.0;47.0;off
2023-07-15 12:57:31.090 [bus error] poll bai PrEnergyCountHwc2 failed: ERR: SYN received
2023-07-15 12:57:32.740 [update notice] received read bai DateTime QQ=10: nosignal;03:24:09;-.-.-;-
2023-07-15 12:57:33.003 [update notice] received poll-read bai Status02 QQ=10: off;60;30.0;100;50.0
2023-07-15 12:57:35.700 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2023-07-15 12:57:37.106 [bus error] poll bai PrEnergyCountHwc3 failed: ERR: SYN received
2023-07-15 12:57:43.310 [update notice] sent poll-read bai PrEnergySumHc1 QQ=31: 4294967164
2023-07-15 12:57:45.744 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2023-07-15 12:57:49.293 [update notice] sent poll-read bai PrEnergySumHc2 QQ=31: 0
2023-07-15 12:57:50.770 [update notice] received read bai Status01 QQ=10: 50.0;48.0;-;37.0;47.0;off
2023-07-15 12:57:52.710 [update notice] received poll-read bai Status02 QQ=10: off;60;30.0;100;50.0
2023-07-15 12:57:55.119 [bus error] poll bai PrEnergySumHc3 failed: ERR: wrong symbol received
2023-07-15 12:57:55.830 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2023-07-15 12:58:00.771 [update notice] received read bai Status01 QQ=10: 50.0;48.0;-;36.0;47.0;off
2023-07-15 12:58:01.310 [update notice] sent poll-read bai PrEnergySumHwc1 QQ=31: 564866919
2023-07-15 12:58:02.750 [update notice] received read bai DateTime QQ=10: nosignal;03:24:40;-.-.-;-
2023-07-15 12:58:05.761 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2023-07-15 12:58:07.300 [update notice] sent poll-read bai PrEnergySumHwc2 QQ=31: 0
2023-07-15 12:58:10.780 [update notice] received read bai Status01 QQ=10: 50.0;48.0;-;36.0;47.0;off
2023-07-15 12:58:13.300 [update notice] sent poll-read bai PrEnergySumHwc3 QQ=31: 0
2023-07-15 12:58:15.774 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2023-07-15 12:58:19.129 [bus error] poll bai PumpHours failed: ERR: wrong symbol received
2023-07-15 12:58:20.763 [update notice] received read bai Status01 QQ=10: 50.0;48.0;-;36.0;47.0;off
2023-07-15 12:58:22.771 [update notice] received update-read broadcast vdatetime QQ=10: 13:02:46;15.07.2023
2023-07-15 12:58:25.089 [bus error] poll bai ReturnTemp failed: ERR: SYN received`

piotr-lwks, 2023-07-22:
I'm seeing the same problem even though the WiFi signal is not in a bad state:

> WiFi station: Nikiel, 192.168.8.128, 58% (-71dBm)

john30, 2023-09-16:
you need to have at least 70% signal strength plus a low latency on your wifi, otherwise it can't work without latency issues

john30, 2023-09-17:
duplicate of #3

closed 2023-09-17 by john30