john30/ebusd-esp32 #7: Can I disable the AP at startup ?

farfade, 2023-07-19:
Hello,

I'm not sure to understand the way the adapter V5 works.

When I start it, the wifi is in AP mode instead of being connected as a client to my router.
I've got to type "sta" in the console to connect.

I've tried "ap --store --off", but it does not work.

What is the way to make the adapter directly connect as a client instead of starting the Access Point ?

Thanks for reading !

john30, 2023-08-06:
when station is configured, easi will start both network mode, station and ap. ap is only run for up to 2 minutes to get a chance to reconfigure the adapter, but this will be removed in the next version

farfade, 2023-08-06:
Many thanks !

closed 2023-09-16 by john30