john30/ebusd-esp32 #8: error reading config files

ThiloDr, 2023-07-22:
Hi John, 

Thanks for the adapter. Looking forward to use it. For some reason I just can't connect the adapter to ebusd. 

/var/log/ebusd.log is saying:
2023-07-22 12:08:58.722 [main notice] ebusd 3.0.595c7c0 started
2023-07-22 12:08:58.722 [main error] error reading config files: ERR: element not found, last error:
2023-07-22 12:08:58.722 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:08:58.723 [bus notice] bus started with own address 31/36
2023-07-22 12:09:08.723 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:09:18.723 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:09:28.723 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:09:38.724 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:09:48.724 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:09:58.724 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:10:08.724 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:10:18.724 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:10:28.725 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:10:38.725 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:10:48.725 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:10:58.726 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:11:08.726 [bus error] unable to open ens:/dev/ttyACM0: ERR: element not found
2023-07-22 12:11:08.832 [main error] update check error: 301 Moved Permanently

/etc/default/ebusd:
EBUSD_OPTS="--scanconfig -d ens:/dev/ttyACM0"

ls -la /dev/ttyACM0 retuns:
crw-rw---- 1 root dialout 166, 0 Jul 22 12:17 /dev/ttyACM0

Any idea? Thanks 
Thilo

ThiloDr, 2023-07-24:
easi> seems to have some connection though:

ebusd connected: yes (inactive)
eBUS signal: acquired

john30, 2023-08-06:
you can't run it with such an old ebusd. 23.1 is the minimum required version

closed 2023-08-06 by john30