john30/ebusd-esp32 #9: No connection to recovair260

cvastag, 2023-08-03:
Try to connect Revovair 260
Adapter ist Version 5

Only Blue and Green LED is on, red LED no activity. Also try to change Cable and firmware of Adapter

easi:
ebusd connected: yes (inactive)
eBUS signal: acquired

Ebusd:

ebusctl info                                                                                                                                                                  
version: ebusd 23.2.23.2                                                                                                                                                      
update check: OK                                                                                                                                                              
device: 10.0.20.245:9999, enhanced, firmware 1.1[3501].1[3501]                                                                                                                
access: *                                                                                                                                                                     
signal: acquired                                                                                                                                                              
symbol rate: 23                                                                                                                                                               
max symbol rate: 32                                                                                                                                                           
scan: finished                                                                                                                                                                
reconnects: 2                                                                                                                                                                 
masters: 1                                                                                                                                                                    
messages: 11                                                                                                                                                                  
conditional: 0                                                                                                                                                                
poll: 0                                                                                                                                                                       
update: 4                                                                                                                                                                     
address 04: slave #25, ebusd                                                                                                                                                  
address ff: master #25, ebusd

john30, 2023-08-06:
looks as if the recovair does not send anything to the bus on its own, otherwise the master/slave addresses would appear

try using full scan option of ebusd

cvastag, 2023-08-06:
With full scan I can find the Recovair

root@ john30-ebusd1: /# ebusctl scan result
1 scan (s) still running
08; Vaillant; recoV; 0227; 0203

But there is no csv in the config folder.

Can you pls upload 08.recov.csv?

john30, 2023-12-23:
wrong repo, so closing

closed 2023-12-23 by john30