john30/ebusd-esp32 #10: where is the source code ? and the binary ?

farfade, 2023-08-06:
Hello,

I notice that this repository is empty.
Is it on purpose that the source code is not released ?

Thanks

john30, 2023-08-15:
yes, that's on purpose.

farfade, 2023-08-15:
Oh 🥺

That's a pity but I respect your choice.

Please juste keep in mind all your users if you stop maintaining it (bumping librairies, adding new protocols support...), especially in case of unplanned events.

Can you release at least the binary directly as you do on the other repository ? I want to have a backup of everything ruling my home but I would save time if I don't have to read the flashed memory.

Best

john30, 2023-08-15:
you can just follow what the browser gets when [installing one of the versions](https://adapter.ebusd.eu/v5/firmware.en.html).

that is the manifest file like [20230815.json](https://adapter.ebusd.eu/v5/files/20230815.json) and then all the files listed in there in the same path.

you could even automate that

closed 2023-09-16 by john30