john30/ebusd-esp32 #11: Passwort für die Webseite erforderlich und Server startet nach Reboot nicht

Streifix, 2023-08-22:
Ich habe heute die HW bekommen und versucht zu konfigurieren. Die Standardadresse für das Webinterface habe ich so umgestellt, dass der ebus-Adapter sich in meinem existierendes WLAN anmeldet und dabei eine fixe IP hat. Zunächst konnte ich über diese IP nicht mit einem Browser zugreifen.
Wenn ich allerdings im seriellen Interface (via USB) den Befehl 'sta' ohne Parameter eingebe, ist die IP über den Browser (http, Port 80) erreichbar.

1. Allerdings wird für den Zugriff von Konfiguration und Log ein Benutzername und Passwort verlangt, das ich nicht kenne.
2. Nach einem Reboot ist die IP wieder nicht erreichbar, bis ich wieder über die 'serielle' Schnittstelle 'sta' eingebe

Was mache ich falsch?

Günter

HeikoGr, 2023-08-23:
Ich hatte zuerst die selben Probleme.
die Firmware vom 15.08. hat die von dir genannten Probleme bei mir behoben:
https://adapter.ebusd.eu/v5/firmware.html

Dort steht auch wie du die firmware nur mit Browser uns USB Kabel flashen kannst.

john30, 2023-08-24:
the fix IP settings seems to be not working (again), need to fix that.

the issue with the empty authentication should be solved with 2023-08-15, but for some browsers, another adaption is needed when auth is desired. will be included in next version

Streifix, 2023-08-26:
Vielen Dank für die Hilfe. Ich dachte, die Version, die ich hatte, sei 'neu genug'. Die Probleme 1 und 3 sind gelöst. Allerdings bleibt das Problem 2 bestehen ("Nach einem Reboot ist die IP wieder nicht erreichbar, bis ich wieder über die 'serielle' Schnittstelle 'sta' eingebe."). Konkret besteht das Problem wohl darin, dass der Adapter sich nicht mit meinem WLAN verbindet, erst wenn ich über die 'serielle' Schnittstelle 'sta' eingebe.

Günter, 23.08.23

Streifix, 2023-08-26:
Dear John,
thank you for your answer. When do you expect the next version?
For me I have to use the fixed IP.
Günter, 26.08.23

john30, 2023-09-03:
a new version was just released and this addresses the fix IP issue as well as the authentication headers for some browsers, see here: https://adapter.ebusd.eu/v5/ChangeLog#20230903

Streifix, 2023-09-07:
Dear John,

thank you for fixing this problem really fast. I flashed the device with the new firmware and it works fine.

However. I am facing other problems, now with the ebusd-daemon (my vaillant system seems to have no interpretable messages), but this is a different topic for a different thread.
Günter, 07.09.23

closed 2023-09-09 by john30