john30/ebusd-esp32 #13: TCP connection over wifi works perfectly but large number of errors over USB serial

dfeist, 2023-09-01:
Just received updated firmware and set up a new v5 board for use with Vaillant AroTherm Plus.

I'm using this config for USB and Wi-Fi alternatives:

```
EBUSD_OPTS="--scanconfig --device=ens:/dev/ttyACM0 -c /etc/ebusd --latency=0"
#BUSD_OPTS="--scanconfig --device=ens:192.168.0.161:9999 -c /etc/ebusd --latency=10"
```

What I find is that when using Wi-Fi everything works great, scanning happens quickly and the config files are determined just as quickly.  Also, when reading values via 'ebusctl' I don't see errors.  I've switched the USB cable but observed the same.

But, when using the USB connection I get numerous errors and scanning takes a long time (and sometimes doesn't succeed).  Also, once csv's files are determined 60% (or more) of reads fail.  What debugging can I do to try to understand the cause f these

The errors I see are as follows:

```
2023-09-01 12:19:15.755 [main debug] performing regular tasks
2023-09-01 12:19:15.755 [bus info] scan 05 cmd: 3105070400
2023-09-01 12:19:15.768 [bus debug] start request 31
2023-09-01 12:19:15.768 [bus debug] arbitration start with 31
2023-09-01 12:19:15.816 [bus debug] arbitration won
2023-09-01 12:19:15.816 [bus debug] arbitration delay 4 micros
2023-09-01 12:19:15.816 [bus info] arbitration delay 4 - 4 micros
2023-09-01 12:19:15.816 [bus debug] switching from ready to send command
2023-09-01 12:19:15.846 [bus debug] notify request: ERR: read timeout
2023-09-01 12:19:15.846 [main error] scan config 05: ERR: read timeout
2023-09-01 12:19:15.846 [bus debug] ERR: read timeout during send command, switching to skip
```

If I increase the `--latency` to `50ms` I no longer see read timeouts but instead get:

```
2023-09-01 12:28:23.694 [bus debug] start request 31
2023-09-01 12:28:23.694 [bus debug] arbitration start with 31
2023-09-01 12:28:24.093 [bus debug] arbitration won
2023-09-01 12:28:24.093 [bus debug] arbitration delay 3 micros
2023-09-01 12:28:24.093 [bus debug] switching from ready to send command
2023-09-01 12:28:24.136 [bus debug] notify request: ERR: SYN received
2023-09-01 12:28:24.136 [bus debug] ERR: SYN received during send command, switching to ready
2023-09-01 12:28:24.136 [bus error] send to 15: ERR: SYN received, retry
2023-09-01 12:28:24.180 [bus debug] start request 31
2023-09-01 12:28:24.180 [bus debug] arbitration start with 31
2023-09-01 12:28:24.228 [bus debug] arbitration won
2023-09-01 12:28:24.229 [bus debug] arbitration delay 5 micros
2023-09-01 12:28:24.229 [bus debug] switching from ready to send command
2023-09-01 12:28:24.272 [bus debug] notify request: ERR: SYN received
2023-09-01 12:28:24.272 [bus debug] ERR: SYN received during send command, switching to ready
2023-09-01 12:28:24.272 [bus error] send to 15: ERR: SYN received
2023-09-01 12:28:24.272 [bus error] send message part 0: ERR: SYN received
2023-09-01 12:28:24.272 [main debug] <<< ERR: SYN received
```

john30, 2023-09-03:
can't reproduce this. are you sure that no other program is using the same USB port? check with `lsof` e.g.

john30, 2023-09-17:
I've found an issue hidden in the ESP32-C3 UART that reports a falsy non-TX-readiness, which might be related to this.

can you check with the [new firmware 20230917](https://adapter.ebusd.eu/v5/ChangeLog#20230917) just released?

dfeist, 2023-09-17:
> can you check with the [new firmware 20230917](https://adapter.ebusd.eu/v5/ChangeLog#20230917) just released?

Massive improvement with the latest firmware.  It's now consistently discovering the relevant config files quickly and is able to read all values (quicker than via wifi) without any timeouts or `SYN received during send command`'s in the log.

john30, 2023-09-18:
great, thanks

closed 2023-09-18 by john30