john30/ebusd-esp32 #15: Error reading config files

frankkaktus, 2023-09-05:
In the previous version (15.8.23) there was a problem with accepting the the configuration of the web interface (WIFI station/client & credentials). The values includes were ignored. 
With the new version (3.9.23) the behaviour is changed:

- the original configured data are active
- the fields in the configuration area are empty 
- I've set the values again and tried to save it, but the box to enter password and credentials is only active for 5s
Mostly I was to slow to enter password and credentials...
One times I was quick enough and have changed the WIFI station/client values. At next request the values were listed, but password and credentials were deleted.
Now I've set password and credentials again and save the changes. Next connection works, but the credentials will not be requested entering the configuration page. After power off cycle the adapter was not possible to reach again...

frankkaktus, 2023-09-06:
After using the erase tool and reflashing the software with active erase option I can activate again the hotspot and configure the adapter. But it does not connect to the WIFI network as before.
What could be done to reactivate the adapter?

frankkaktus, 2023-09-08:
Here again the current status:
After erasing, flashing the newest software and entering the network data again it works as expected. Thanks John for the good work...
Open stays from my side of view:
- the handling of the configured data (used, but not viewed or possible to edit).
- the only 5s view of the password dialog after flashing the new software
Sorry, the rest of the behaviour could be located in front of screen...
Frank

john30, 2023-09-09:
not sure what you mean by "handling of the configured data". which values do you mean exactly?

I also don't get the point of the 5s credentials dialog. please explain how that could be reproduced or describe exactly the steps that you did. maybe also a hint to which browser you're using would be helpful

frankkaktus, 2023-09-10:
> "not sure what you mean by "handling of the configured data". which values do you mean exactly?"
The adapter was delivered with the 15.8.23-version (I hab proved). I had configured the "WIFI station/client" configuration and the credentials for "httpd" - and the data were ignored. After installing the 3.9.23-version and power off cycle, the data were used. Instead of DHCP the IP address and the name switch to the configured values. Entering the configuration dialog all the configured and used data are not shown. It seems, there are two levels of data storage...

> "I also don't get the point of the 5s credentials dialog. please explain how that could be reproduced or describe exactly the steps that you did. maybe also a hint to which browser you're using would be helpful"
In this status I entered the "WIFI station/client" data again, not the credentials for the "httpd". As the credentials were entered in the old version, a box for entering user/password was opened. But after 5s it was closed, mostly to quickly to enter both data.
The used browser was firefox under Linux.

john30, 2023-09-16:
I still don't get your points.

anyway, I've increased the API call timeout in the UI, so maybe that eases it. check out the new version once it is released

john30, 2023-12-23:
closed for inactivity

closed 2023-12-23 by john30