john30/ebusd-esp32 #19: Network works only with WIFI/client on

Dennis1000, 2023-09-14:
I've connected to the adapter via WIFI "ebus", configured both WIFI client (dynamic IP) AND network (using a USR-ES1) with static IP address and it works. But as soon as I disable the WIFI connection and reboot, the adapter is pingable on the fixed network address but the UI doesn't show up.
Enabling the WIFI client (starting from scratch via "ebus" WIFI) and the enabling the network, the UI shows up on both IPs (network and WIFI).

john30, 2023-09-16:
good catch! solved in next version

john30, 2023-09-17:
could you check with the [new firmware 20230917](https://adapter.ebusd.eu/v5/ChangeLog#20230917) just released?

Dennis1000, 2023-09-17:
Works perfectly. Disabled WIFI/client, did a reboot and the UI shows up! Thanks!

john30, 2023-09-18:
great, thanks!

closed 2023-09-18 by john30