john30/ebusd-esp32 #20: Update firmware by webUI

pepsonEL, 2023-09-18:
Hi
Is any chance to implement to this version option update FW by WebUI (not by wire) ?

john30, 2023-09-23:
when I have time to and it fits in the available room with ebusd

VivantSenior, 2023-12-20:
@john3, were you able to implement OTA from WebUI in ESP32-C3 rev 4? I found information on [this page](https://adapter.ebusd.eu/v5/steps.en.html#adapterflash):
`Newer firmware versions can be easily updated via OTA directly on the adapter website.`
but was unable to find it in the WebUI:
[image]

VivantSenior, 2023-12-20:
After [manual update](https://adapter.ebusd.eu/v5/firmware.en#flash) to 20231217 I was able to find "Update firmware" button:
[image]

john30, 2023-12-23:
yes

closed 2023-12-23 by john30