john30/ebusd-esp32 #21: Adpter v5 - problem with USR-ES1 module

zyzakin, 2023-09-18:
I have version 5 of the module, and I have to use an Ethernet connection because I am not getting a signal from the router via WiFi. Besides, I prefer a wired connection (as more reliable) than a radio connection.
Unfortunately, after connecting the module and setting the correct configuration [image], the module is not detected (it has the status not available).
After repeated attempts to reset and change the WiFi configuration ( switching on, off ) finally the module was "noticed" and it started to work somehow. The USR-ES1 module I am using is HanRun HR961160C 2315.
Unfortunately today it stopped and I decided it was best to report it.

john30, 2023-09-23:
after pressing the quick-settings button, the device needs to be restarted as well to run the module detection. did you do that?
anyway, is it the up-to-date firmware version? if not, then please update first.
the status shown on top of the web UI would be important to know as well

zyzakin, 2023-09-27:
I just received the new module and the situation is as before.

To answer your question, yes, I did a reboot repeatedly, but it had no effect. I even turned off the power - to no avail.

Finally, I changed the location of the module and connected via WiFi and it works. Ultimately, however, I would like the connection to be via Ethernet.

Paste a snapshot of the status

[image]
[image]
I am worried about the increase in the temperature of the module
[image]

This time it shows a little more in the logs, but I can't interpret it.

`

17:17:33	
	network trigger installed
17:17:33	
Info	replExecuteFromFile: executing command: on -o server 2 "after -o -m 500 0 restart"
17:17:33	
Info	executeCommand: executing command: on -o server 2 "after -o -m 500 0 restart"
17:17:33	
	server trigger installed
17:17:33	
Info	replExecuteFromFile: executing command: if "isnvs sta.on" sta
17:17:33	
Info	executeCommand: executing command: if "isnvs sta.on" sta
17:17:33	
Info	executeCommand: executing command: isnvs sta.on
17:17:33	
Info	executeCommand: executing command: sta
17:17:33	
Info	pp: pp rom version: 9387209
17:17:33	
Info	net80211: net80211 rom version: 9387209
17:17:33	
Info	wifi_init: tcpip mbox: 32
17:17:33	
Info	wifi_init: udp mbox: 6
17:17:33	
Info	wifi_init: tcp mbox: 6
17:17:33	
Info	wifi_init: tcp tx win: 5744
17:17:33	
Info	wifi_init: tcp rx win: 5744
17:17:33	
Info	wifi_init: tcp mss: 1440
17:17:33	
Info	wifi_init: WiFi IRAM OP enabled
17:17:33	
Info	wifi_init: WiFi RX IRAM OP enabled
17:17:33	
Info	wifi_init: LWIP IRAM OP enabled
17:17:33	
Info	phy_init: phy_version 1110,9c20f0a,Jul 27 2023,10:42:54
17:17:33	
	using individual IP: 192.168.1.25 / 255.255.255.0, gateway 192.168.1.10
17:17:33	
	sta connecting to 'glabnet'
17:17:35	
Info	cmd_wifi: sta connect, set fix ip
17:17:35	
Info	repl_event_task: wifi event: 1
17:17:35	
Info	executeCommand: executing command: led 0 50
17:17:35	
Info	executeCommand: executing command: start
17:17:35	
[... 180 lines cut ...]
Info	init_ebus: timer started
17:17:35	
Info	executeCommand: executing command: led 0 50
17:17:35	
Info	executeCommand: executing command: exit
17:17:35	
Info	executeCommand: command exit
17:17:35	
Info	executeCommand: command exit
17:17:35	
Info	replExecuteFromFile: successfully executed 45/45 commands from '/data/main.eas'
17:17:35	
	Welcome to eBUS Adapter Shield Interface! Type 'help' to get the list of commands.
17:17:35	
	Your terminal application does not support escape sequences.
17:17:35	
Info	main_task: Returned from app_main()
17:17:35	
Info	statusLedTimerFn: duty set to 50
17:17:40	
Info	host_task_tcp: accepted: 192.168.1.88
17:22:28	
Info	temperature_sensor: Range [-10°C ~ 80°C], error < 1°C

`

john30, 2023-09-29:
the logs shows that the Ethernet module can't be detected:
> Error	w5500.mac: W5500 version mismatched, expected 0x04, got 0x00

are you sure that you plugged it in correctly as shown on the UI?

[image]

zyzakin, 2023-09-30:
I'm sure I've put it in right, because all the wires are on the same side.

I'm pasting a portion of the logs again.

```
13:09:15	Info	gpio: GPIO[6]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
13:09:15	Info	gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
13:09:15	Info	gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
13:09:15	Info	gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
13:09:15	Ethernet not available
13:09:15	    Warning	executeCommand: command failed: eth -t = 1
13:09:15	Info	executeCommand: executing command: if "eth -wt" "var eth 2"
13:09:15	Info	executeCommand: executing command: eth -wt
13:09:15	Error	w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
13:09:15	Error	w5500.mac: emac_w5500_init(715): vefiry chip ID failed
13:09:15	Info	gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
13:09:15	Error	esp_eth: esp_eth_driver_install(228): init mac failed
13:09:15	Error	initialise_ethernet: initialise_ethernet(248): install driver
13:09:15	Error	esp_eth: esp_eth_driver_uninstall(256): invalid argument
13:09:15	Info	gpio: GPIO[6]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
13:09:15	Info	gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
13:09:15	Info	gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
13:09:15	Info	gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
13:09:15		Ethernet not available
13:09:15	Warning	executeCommand: command failed: eth -wt = 1
13:09:15	Info	replExecuteFromFile: executing command: io -a
13:09:15	Info	executeCommand: executing command: io -a
13:09:15	Info	replExecuteFromFile: executing command: function chkethcfg "if \"eth -c\" \"var eth 1\""
13:09:15	Info	executeCommand: executing command: function chkethcfg "if \"eth -c\" \"var eth 1\""
13:09:15		function installed
```
I'm still concerned that the module starts to heat up after attaching the USR-ES1.

Maybe it is a matter of a different model of USR-ES1 ?
[image]

There is another possibility - maybe I damaged something on the module ( temperature rise problem ) ? However, I do not know how to check it

At the moment I'm using over WiFi and if there's no other way, I'll leave it that way. If nothing else can be done, just close this issue.

By the way, it would be nice if when you select the appropriate button it is already highlighted. Thanks to this, the user would be sure what was selected.
Of course, under certain conditions. I can imagine a scenario where the user selects some pre-configuration and then manually changes the configuration of one or two pins, then of course the highlighting disappears.
[image]

Thank you very much for your help so far

john30, 2023-09-30:
having the wires on the same side does not necessarily mean that the USR was plugged in correctly. it needs to have the exact position in the headers as shown.
what is the status of the LEDs on the USR?
the log again tells that the module is not recognized.

zyzakin, 2023-10-01:
photo with connected module.
[image]

Nothing has changed in the logs.

```
18:48:13	Info	replExecuteFromFile: executing command: var eth 0
18:48:13	Info	executeCommand: executing command: var eth 0
18:48:13	Info	replExecuteFromFile: executing command: if "eth -t" "var eth 1" else "if \"eth -wt\" \"var eth 2\""
18:48:13	Info	executeCommand: executing command: if "eth -t" "var eth 1" else "if \"eth -wt\" \"var eth 2\""
18:48:13	Info	executeCommand: executing command: eth -t
18:48:13	Info	statusLedTimerFn: duty set to 10
18:48:13	Error	w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
18:48:13	Error	w5500.mac: emac_w5500_init(715): vefiry chip ID failed
18:48:13	Info	gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
18:48:13	Error	esp_eth: esp_eth_driver_install(228): init mac failed
18:48:13	Error	initialise_ethernet: initialise_ethernet(248): install driver
18:48:13	Error	esp_eth: esp_eth_driver_uninstall(256): invalid argument
18:48:13	Info	gpio: GPIO[6]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
18:48:13	Info	gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
18:48:13	Info	gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
18:48:13	Info	gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0

```

john30, 2023-10-01:
you want to look at the bottom of the USR module and check the numbering on the Wiznet chip

zyzakin, 2023-10-01:
I have two types of W5500 modules. It gives the numbers printed on the chips

- DSW618/2213 (HR961160C/2315)
- DJ4828/2236 (HR961160C/2321)

john30, 2023-10-03:
do they really read W5500 on it? because any other chip is not supported

and what about the LEDs I've asked?

zyzakin, 2023-10-03:
I don't know which LEDs on the USR-ES1 module you are asking about ? I took a picture where you can see the green (Link) and orange (Active) LED of the LAN connector and the red color of the power LED below. The other LEDs on the USR-ES1 are not visible.

I also pasted a photo of the module itself.
[image]

john30, 2023-10-08:
do you know if the modules are working? We've seen many items that are just broken even though the LEDs are shining.

anyway, to ensure that there is no connectivity issue from adapter to USR module, you can use a short wire to acknowledge that all IO pins to the module are OK.
In order to to that, remove the module from the adapter and then use the adapter UI to set the pin configuration to "+all in" and apply that.
then connect the wire to the IO pins against a GND pin one by one. if connected to GND, the UI should show the level as "L" (after pressing the reload Icon in the "Status" tab header).
you could also check the opposite (i.e. set an individual IO pin to output H/L and check it with another IO pin set to input).
if thats OK, and you're USR module still does not work, it is broken to 99%.

john30, 2023-12-23:
closed for inactivity

closed 2023-12-23 by john30

dlbogdan, 2024-03-07:
I have the same issue with a (probably clone) W5500 chinezium pcb. 
I think I saw on a forum on esp-idf that these clones need 100ms delay before polling the ID or something. 

nevermind, found it here: https://github.com/espressif/esp-idf/issues/11331