john30/ebusd-esp32 #25: After Starting Ebusd no further Master and Slaves found

ChrisLan12, 2023-10-02:
Hello,
I've started the ebusd-deamon with the following options:
EBUSD_OPTS="--device=IP-ADRESSE:9999 --scanconfig --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/de/ --receivetimeout=100000 --latency=50"
The command ebusctl i brings the following result:
version: ebusd 22.4.v22.4
update check: version 23.2 available
device: 192.168.0.112:9999
signal: acquired
symbol rate: 46
max symbol rate: 139
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
and the Log-File-entry is:
2023-10-02 18:57:49.859 [main notice] ebusd 22.4.v22.4 started with auto scan on device IP-Adress:9999
2023-10-02 18:57:49.898 [bus notice] bus started with own address 31/36
2023-10-02 18:57:49.914 [bus notice] signal acquired
2023-10-02 18:58:09.036 [bus notice] max. symbols per second: 108
2023-10-02 18:58:39.005 [bus notice] max. symbols per second: 112
2023-10-02 18:58:40.025 [bus notice] max. symbols per second: 114
2023-10-02 18:59:40.005 [bus notice] max. symbols per second: 115
2023-10-02 18:59:56.518 [main notice] update check: version 23.2 available
2023-10-02 19:00:40.005 [bus notice] max. symbols per second: 119
2023-10-02 19:01:41.036 [bus notice] max. symbols per second: 131
2023-10-02 19:02:41.007 [bus notice] max. symbols per second: 137

No further Master oder Slaves where found

The WIFI-Signal is 100%. The Ebus-Line ist less than 1 meter

Greetings
Christoph

ChrisLan12, 2023-10-02:
Now, a further Master #2, address 10 has appeared

john30, 2023-10-03:
the options to ebusd would be correct only if you had configured the adapter to non-enhanced protocol.
please read the instructions (getting tired of answering the same question over and over again): https://adapter.ebusd.eu/v5/index.en.html#wifi
> The ebusd device configuration is e.g. -d ens:192.168.178.2:9999, where 192.168.178.2 has to be replaced with the right IP address.

duplicate of #24

closed 2023-10-03 by john30