john30/ebusd-esp32 #28: [network error] certificate verify failed

oluschka99, 2023-10-19:
Hello,
my ebus adapter is online, but i can not run the docker image, get the following error message:

  pi@raspberrypi:~ $ docker run -it --rm -p 8888 john30/ebusd --scanconfig --device=192.168.178.46:9999
  1931-02-09 00:12:53.034 [network error] HTTP connect: 337047686=certificate verify failed
  1931-02-09 00:12:53.034 [network error] HTTP new_ssl_conn: 336462231=shutdown while in init
  1931-01-28 10:33:20.-1097 [main error] invalid configPath URL (connect)

[image]

Thank you very mush

john30, 2023-10-20:
please try with the :devel image. I recently discovered that the ssl library can go into a weird irreparable status somehow and the :devel image curimvents this

oluschka99, 2023-10-21:
Hello,
I have done it this way, without success:
    _pi@raspberrypi:~ $ docker run --rm -it -p 8888 john30/ebusd:devel --scanconfig -d 192.168.178.46:9999 --latency=20 --mqttport=1883 --mqtthost=192.168.178.50
    1931-02-17 19:50:09.009 [network error] HTTP connect: 337047686=certificate verify failed
    1931-02-17 19:50:09.009 [network error] HTTP new_ssl_conn: 336462231=shutdown while in init
    19_31-02-06 06:10:56.-1095 [main error] invalid configPath URL (connect)_

Then so, but Container did not run:
    pi@raspberrypi:~ $ docker run -d --name=ebusd -p 8888 john30/ebusd:devel --scanconfig -d 192.168.178.46:9999 --mqttport=1883 --mqtthost=192.168.178.50
    bf2c669e33db029d0dbb32382e11c6626cfea62b41177da3b370e04ef9a31e52

Then I installed a container via Portainer, container runs but no communication with adapter.
In the end I installed ebusd-debian and everything works. I will let it run like this.

Thanks a lot

john30, 2023-10-22:
next step is to increase the logging, so that we know what exactly ssl lib is complaining about

closed 2023-10-31 by oluschka99