john30/ebusd-esp32 #29: eBus Adapter info into ebusd

Koky05, 2023-10-20:
Would it be possible to populate some eBus Adapeter info into ebusd and then read them in KNX / HA?
For example WiFi strength or FW version?

john30, 2023-10-20:
firmware version is already reported to HA and reflected as device update check entity.
wifi strength might be interesting to add indeed

JavanXD, 2024-05-03:
For HA there is already the option to fetch the data via API request from your Adaper:

The in-built MQTT option which got recently released did not work unfortunately, as it does comes with a few bugs, so I helped myself:

```yaml
sensor:
- platform: rest
  name: "eBUS Adapter JSON" 
  resource: http://192.168.178.XX/api/v1/status
  json_attributes_path: "ebus"
  value_template: "{{ value_json.ebus.signal }}"
  json_attributes:
    - on
    - connected
    - proto
    - host
    - signal
  headers:
    Content-Type: application/json
  scan_interval: 60
```

Other values can be fetched too, by changing the json_attributes_path.

john30, 2024-05-05:
being a bit more specific wrt "comes with a few bugs" would certainly be helpful...