john30/ebusd-esp32 #31: Massive Störung auf Frequenzband 868MHz (HomeMatic)

titus947, 2023-10-26:
Betreibe den kürzlich bestellten bus adapter 5 per USB an einer Raspberry Pi 4 mit Homematic Funkmodul und ConBee 2 USB Sender um Daten aus einer Vaillant WP auszulesen. Das Auslesen scheint auch ganz gut zu klappen aber sobald ich den Adapter anstecke stört er dermaßen das von Homematic verwendete Frequenzband (868MHz), dass das gesamte HomeMatic Funknetz ausfällt und HM Geräte redetet werden müssen. das WLAN Modul ausschalten hat leider auch nichts gebracht. Ist dieses Problem bekannt und wenn ja, gibt es irgendeine Möglichkeit, das Störsignal auszuschalten? Ich verdächtige das WLAN Modul aber das war eigentlich auch das erste, was ich ausgeschaltet habe. Bin für jeden Hinweis dankbar, denn so lässt sich der Adapter bei mir leider nicht einsetzen.

VG

PhilippEngler, 2023-10-30:
Wenn ich das richtig gelesen und verstanden habe, betreibst Du das HM-Funkmodul sowie Deinen ZigBee-Stick auf einem RaspberryPi 4. Hast Du den eBus-Adapter an einen der blauen USB-Ports angeschlossen?

Ich glaube nicht, dass der eBus-Adapter das Problem ist, er löst es lediglich aus. Falls noch nicht getan, ließ Dich mal bitte im HomeMatic-Forum in die Probleme des RaspberryPi 4 mit HomeMatic ein ([ein Artikel](https://homematic-forum.de/forum/viewtopic.php?f=31&t=65293)). Dort sind auch einige Lösungsmöglichkeiten aufgezeigt.

Viele Grüße

Philipp

titus947, 2023-11-15:
Scheint tatsächlich daran zu liegen. Ärgerlich, auf der Pi4 ist pivCCU, zigbee und iobroker drauf, gerade letzteres braucht schon ordentlich RAM, das war der Grund dafür, die Pi4 anzuschaffen.

Danke für die Infos!

VG

closed 2023-12-23 by john30