john30/ebusd-esp32 #32: Keine Verbindun tu

burni737, 2023-11-10:


burni737, 2023-11-10:
Ich versuche nun seit einigen Wochen meinen ebus adapter v5 mit meiner Vaillant Heizung in Betrieb zu nehmen. Leider ohne Erfolg.

Firmware auf dem ebus-adapter: 2023-10-15
Ebusd version: ebusd 23.2.23.2-52-gb9012326

Vaillant-Komponenten: 
- Vaillant auromatic 560
- Vaillant ecoTec Plus mit calorMATIC VRC 470/4

Bei der auromatic verwende ich die beiden mittleren Pins des RJ11 Buchse und bei der Gasheizung die Bus-Klemmen auf der Platine. Ich habe beide Bus-Leitungen gemessen.

Bei der auromatic 560 habe ich konstant ca. 22 Volt
[image]

Bei der Gasheizung sehe ich ein Rechtecksignal.
[image]

ebusd wird mit folgenden Parametern gestartet: --scanconfig=full --device ens:/dev/ttyACM0 --loglevel=debug -p 8888 -l /var/log/ebusd.log

Die grüne LED auf der Platine leuchtet, sonst tut sich aber nicht viel. "ebusctl info" gibt mir die folgenden Werte aus;
version: ebusd 23.2.23.2-52-gb9012326
device: /dev/ttyACM0, enhanced high speed
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

Da ich aktuell keine Ahnung mehr habe, woran es liegen könnte, wäre ich für jeden Tipp dankbar.

john30, 2023-11-18:
only the one with the rectangular signal seems to be working out of the box. so when you connect the adapter to that one, you should see at least a signal acquired on ebusd. you can then try to scan for devices using a full scan or broadcast scan e.g.

for the other one, you can configure ebusd to act as SYN generator (--generatesyn) so that communication is possible at all. then you can again check with full/broadcast scan what devices identify themselves.

connecting two busses is possibly in principle, but I wouldn't dare to (I have 3 separate ones myself). anyway, if you do, watch the polarization and disconnect if anything gets warmer thatn ususal (as it might consume extra current which it shouldn't). but as said, I wouldn't.

burni737, 2023-11-23:
Thank you for your reply.

Today I had time to test again. 

For the one with the rectangular signal, I do not get a "signal acquired", no matter what I do. I did even test different modes of operation, 

```
ebusd -f --device /dev/ttyACM0
ebusd -f --device enh:/dev/ttyACM0
ebusd -f --device ens:/dev/ttyACM0
```

but I always have "status: no signal".

For the other one, I tested with --generatesyn, but then I get a lot of bus errors:

```
ebusd -f --device ens:/dev/ttyACM0 --generatesyn
2023-11-23 20:05:02.032 [bus notice] device status: resetting
2023-11-23 20:05:02.032 [main notice] ebusd 23.2.23.2-52-gb9012326 started with broadcast scan on device: /dev/ttyACM0, enhanced high speed
2023-11-23 20:05:02.640 [bus notice] bus started with own address 31/36
2023-11-23 20:05:02.640 [bus error] device status: unexpected enhanced byte 2
2023-11-23 20:05:02.641 [bus error] device status: unexpected available enhanced following byte 1
2023-11-23 20:05:02.673 [bus notice] device status: reset, supports info
2023-11-23 20:05:02.673 [bus error] device status: unexpected enhanced byte 2
2023-11-23 20:05:02.673 [bus error] device status: unexpected available enhanced following byte 1
2023-11-23 20:05:02.673 [bus notice] acting as AUTO-SYN generator
2023-11-23 20:05:02.673 [bus notice] signal acquired
2023-11-23 20:05:02.674 [bus error] device status: unexpected enhanced byte 2
2023-11-23 20:05:02.674 [bus error] device status: unexpected available enhanced following byte 1
```

I also tested different operation modes:

```
ebusd -f --device /dev/ttyACM0 --generatesyn
2023-11-23 20:06:00.588 [main notice] ebusd 23.2.23.2-52-gb9012326 started with broadcast scan on device: /dev/ttyACM0
2023-11-23 20:06:01.197 [bus notice] bus started with own address 31/36
2023-11-23 20:06:01.624 [bus error] received 81 instead of AUTO-SYN symbol
2023-11-23 20:06:01.624 [bus notice] signal acquired
2023-11-23 20:06:03.457 [bus notice] acting as AUTO-SYN generator
2023-11-23 20:06:03.457 [bus notice] signal acquired
2023-11-23 20:06:11.198 [main notice] starting initial broadcast scan
2023-11-23 20:06:11.271 [bus error] send to fe: ERR: wrong symbol received, retry
2023-11-23 20:06:11.334 [bus error] received fe instead of AUTO-SYN symbol
2023-11-23 20:06:11.334 [bus notice] signal acquired
2023-11-23 20:06:11.334 [bus error] send to fe: ERR: no signal, give up
2023-11-23 20:06:11.334 [main error] initial scan failed: ERR: no signal
2023-11-23 20:06:20.700 [main notice] starting initial broadcast scan
2023-11-23 20:06:20.852 [bus error] received 31 instead of AUTO-SYN symbol
2023-11-23 20:06:20.852 [bus notice] signal acquired
2023-11-23 20:06:20.852 [bus error] send to fe: ERR: no signal, give up
2023-11-23 20:06:20.852 [main error] initial scan failed: ERR: no signal
2023-11-23 20:06:20.952 [bus error] received 31 instead of AUTO-SYN symbol
2023-11-23 20:06:20.952 [bus notice] signal acquired
```

What would be the best way to further debug this issue? 
If there anything I can meassure or test to see if the adapter is "broken"?

john30, 2023-12-17:
did you set up the adapter to use the enhanced protocol and to connect on the Raspi lines? or how is the host running ebusd connected to the adapter?

burni737, 2024-01-05:
I erased all my settings and uploaded the latest firmware to the adapter. This time I entered my wlan credentials to connect to wlan. I used the web interface to change the configuration to enhanced protocol and usb port. After reboot I verified my settings in the web interface and everything was fine.

After connecting the ebus my heating system was detected and ebus communication seams to work fine now. 

After all I am not sure why it is working now. I assume I did not configure the adapter correctly via console.

Many thanks for your help.

closed 2024-01-05 by burni737

Solarer, 2024-01-11:
@burni737 
Did you initially connect your adapter via USB to the raspberry? I am facing the exact same issues with my Vaillant devices as you and also managed to get it working via WIFi...²
I will open a new ticket and would like to link to this ticket aswell

burni737, 2024-01-12:
@Solarer 
The adapter was and is connected via usb. Initially I did not plan to use wifi, therefore I did not configure it. I tried to configure the adapter via the command line interface but I did not get it working. Therefore I tryed something else. I reflashed the firmware and configured the wifi to be able to access the web interface. I had to try it several times but I did get the adapter connected to the network and was able to open the web interface. Via the web interface I changed the config to use enhanced mode and usb. After that, the adapter started working fine. I assume I did something wrong configuring the adapter via the command line interface.

Solarer, 2024-01-12:
Ah, ok. Thank you for the detailed explanation. I will try that as well. I also had issues applying the wifi settings via web interface - saving did not work a few times.
Will post an update once I tested it

Solarer, 2024-01-13:
Also managed to get it working somehow via USB. Thanks again!
I will try to reproduce the steps before opening a new ticket