john30/ebusd-esp32 #34: Stromausfall, v5 Adapter,  hwc OperatingMode ERR: SYN received

cnu80, 2023-11-27:
Hallo, ich betreibe den v5 Adapter an einer Vaillant Wärmepumpe. Das Ganze ist via WIFI mit MQTT an HA angebunden. Funktionierte seit mehreren Wochen ganz gut.

Nach einem Stromausfall funktionierte der Adapter nicht mehr richtig. Bei einem:
```
ebusctl w -c hwc OperatingMode auto
```
kommt nur 
```
ERR: SYN received
```
zurück. Vorher ging es ohne Probleme. (ebusctl w -c mc OperatingMode auto, funktioniert wie gehabt.)

Einmaliges Warmwasser funktioniert (zum Glück) noch.
```
ebusctl w -c hwc load on
done
```

Aber wie bekomme ich mein OperatingMode für hwc zurück? Wo fange ich mit der Fehlersuche an? 

Ein paar Daten:

ebusctl
```
ebusctl find r | grep OperatingMode
hwc OperatingMode = no data stored
mc OperatingMode = on
```
```
ebusctl scan result
08;Vaillant;EHP00;0344;7201
15;Vaillant;UIH00;0226;6901
25;Vaillant;EHP00;0344;7201
50;Vaillant;EHP00;0344;7201
```
```
ebusctl info
version: ebusd 23.2.23.2
update check: OK, device firmware 1[3a0f] available
device: 10.30.0.70:9999, enhanced, firmware 1.8[0139].15[0139]
access: *
signal: acquired
symbol rate: 21
max symbol rate: 127
min arbitration micros: 0
max arbitration micros: 350
min symbol latency: 10
max symbol latency: 60
scan: finished
reconnects: 3
masters: 4
messages: 533
conditional: 194
poll: 96
update: 60
address 00: master #1
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0344;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0226;HW=6901"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0344;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0344;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0344;HW=7201", loaded "vaillant/50.ehp.mc.csv"
```

ebusd-Log
```
2023-11-27 07:39:22.312 [update notice] received read ehp Status QQ=10: 36.38;1.889;0.000;heat;2c
2023-11-27 07:39:22.526 [update notice] received read ehp Status02 QQ=10: disabled;0;100.0;0;100.0
2023-11-27 07:39:22.935 [update notice] received write ehp EnergyBalancingRelease QQ=10: on
2023-11-27 07:39:24.401 [update notice] received poll-read hwc Mode QQ=10: 38;auto;disabled;hwc;00;4
2023-11-27 07:39:24.577 [update notice] received poll-read hwc Status QQ=10: 0;on;36.38;52
2023-11-27 07:39:25.181 [bus error] poll ehp DeltaTempT6T7 failed: ERR: wrong symbol received
2023-11-27 07:39:28.380 [update notice] received update-read broadcast hwcStatus QQ=10: off;36;off
2023-11-27 07:39:30.449 [update notice] received unknown MS cmd: 1025b505072b000100000000 / 00
2023-11-27 07:39:30.932 [update notice] received read ehp Status QQ=10: 36.19;1.893;0.000;heat;2c
2023-11-27 07:39:31.116 [update notice] received read ehp Status02 QQ=10: disabled;0;100.0;0;100.0
2023-11-27 07:39:32.180 [update notice] received read mc DateTime QQ=10: valid;08:39:32;27.11.2023;0.562
2023-11-27 07:39:32.190 [main error] scan config 05: ERR: SYN received
2023-11-27 07:39:32.757 [update notice] received write ehp EnergyBalancingRelease QQ=10: on
2023-11-27 07:39:34.473 [update notice] received poll-read cc Mode QQ=10: 23;auto;disabled;circulation;00;night
2023-11-27 07:39:34.614 [update notice] received poll-read cc Status QQ=10: 0;off;-;0
2023-11-27 07:39:34.788 [update notice] received update-read ehp FlowTemp QQ=10: 33.38;ok
2023-11-27 07:39:36.134 [bus error] poll ehp FlowtempCoolingMin failed: ERR: SYN received
2023-11-27 07:39:38.384 [update notice] received read ehp currenterror QQ=10: 580;-;-;-;-
2023-11-27 07:39:38.526 [update notice] received update-read broadcast hwcStatus QQ=10: off;36;off
2023-11-27 07:39:38.978 [update notice] received update-read ehp SourceTempInput QQ=10: 3.88;ok
2023-11-27 07:39:39.140 [update notice] received update-read ehp ActualEnvironmentPowerPercentage QQ=10: 100
2023-11-27 07:39:39.307 [update notice] received update-read ehp ActualEnvironmentPower QQ=10: 7
2023-11-27 07:39:40.443 [update notice] received unknown MS cmd: 1023b505072b000100000000 / 00
2023-11-27 07:39:41.028 [update notice] received read ehp Status QQ=10: 35.81;1.890;0.000;heat;2c
2023-11-27 07:39:42.315 [update notice] received read ehp Status02 QQ=10: disabled;0;100.0;0;100.0
2023-11-27 07:39:42.553 [update notice] received read ehp errorhistory QQ=10: 1;2;19:25;26.11.2023;580
2023-11-27 07:39:42.967 [update notice] received write ehp EnergyBalancingRelease QQ=10: on
2023-11-27 07:39:44.435 [update notice] received read mc Mode QQ=10: 20;on;0;0;disabled;pool;night
2023-11-27 07:39:44.602 [update notice] received read mc Status QQ=10: 30;off;33.31;20
2023-11-27 07:39:46.505 [update notice] received write mc RoomTempOffset QQ=10: 0.00
2023-11-27 07:39:47.176 [bus error] poll ehp SourcePumpPrerunTime failed: ERR: SYN received
2023-11-27 07:39:48.420 [update notice] received update-read broadcast hwcStatus QQ=10: off;35;off
2023-11-27 07:39:50.574 [update notice] received unknown MS cmd: 1050b505072b000100000000 / 00
2023-11-27 07:39:51.077 [update notice] received read ehp Status QQ=10: 35.44;1.891;0.000;heat;2c
2023-11-27 07:39:51.260 [update notice] received read ehp Status02 QQ=10: disabled;0;100.0;0;100.0
2023-11-27 07:39:52.079 [update notice] received read ehp errorhistory QQ=10: 2;2;16:07;26.11.2023;580
2023-11-27 07:39:52.688 [update notice] received write ehp EnergyBalancingRelease QQ=10: on
2023-11-27 07:39:52.904 [update notice] received update-read ehp FlowTemp QQ=10: 33.19;ok
2023-11-27 07:39:54.509 [update notice] received poll-read hwc Mode QQ=10: 38;auto;disabled;hwc;00;4
2023-11-27 07:39:54.733 [update notice] received poll-read hwc Status QQ=10: 0;on;35.44;52
2023-11-27 07:39:56.478 [update notice] received update-read ehp TimeOfNextPredictedPowerCut QQ=10: -:-:-
2023-11-27 07:39:58.124 [bus error] poll ehp SourceTempOutput failed: ERR: SYN received
2023-11-27 07:39:58.519 [update notice] received write ehp PowerCutPreloading QQ=10: off
2023-11-27 07:39:59.066 [update notice] received update-read broadcast hwcStatus QQ=10: off;35;off
2023-11-27 07:40:00.467 [update notice] received unknown MS cmd: 1025b505072b000100000000 / 00
2023-11-27 07:40:00.813 [update notice] received read ehp Status02 QQ=10: disabled;0;100.0;0;100.0
2023-11-27 07:40:01.013 [update notice] received update-read ehp SourceTempInput QQ=10: 3.88;ok
2023-11-27 07:40:01.140 [update notice] received update-read ehp ActualEnvironmentPowerPercentage QQ=10: 100
2023-11-27 07:40:01.313 [update notice] received update-read ehp ActualEnvironmentPower QQ=10: 7
2023-11-27 07:40:02.160 [update notice] received read ehp errorhistory QQ=10: 3;2;15:48;26.11.2023;580
2023-11-27 07:40:02.641 [update notice] received read mc DateTime QQ=10: valid;08:40:03;27.11.2023;0.562
2023-11-27 07:40:03.015 [update notice] received write ehp EnergyBalancingRelease QQ=10: on
2023-11-27 07:40:04.475 [update notice] received poll-read cc Mode QQ=10: 23;auto;disabled;circulation;00;night
2023-11-27 07:40:04.664 [update notice] received poll-read cc Status QQ=10: 0;off;-;0
2023-11-27 07:40:08.412 [update notice] received update-read broadcast hwcStatus QQ=10: off;35;off
2023-11-27 07:40:09.175 [bus error] poll ehp SourceTempOutputOffset failed: ERR: SYN received
```
v5 Log:
[image]

Danke

john30, 2023-12-17:
try to grab the sent/received bytes using a raw bytes log (ebusctl raw bytes), then I can check what's going on

cnu80, 2023-12-27:
I updated to the newest firmware and after one day I have again connection problems. I saved the output with raw bytes. 
[forgithub.log](https://github.com/john30/ebusd-esp32/files/13777910/forgithub.log)

thank you for your time.

john30, 2023-12-28:
thanks for the log. it reveals delays of 40ms and more and this is way too much for eBUS protocol.
anyway, please update ebusd to the newest version and check if that helps

john30, 2024-04-20:
you might want to try the [newest version](https://adapter.ebusd.eu/v5/ChangeLog#20240420) just released as there were tons of commits related to wifi in ESP-IDF again

cnu80, 2024-04-20:
Thanks for the info. I tried it out and it is working. !!

closed 2024-04-20 by cnu80