john30/ebusd-esp32 #36: Help with setup with Home Assistant and Ariston Genus One

mihaimdinca, 2023-12-18:
Hello! I recently bought an ESP32 based ebusd adapter and have difficulty setting it up. My aim is to use it with Home Assistant (RPi 4B 8Gb) and an Ariston Genus One + Cube S Net. Si far I completed the following:
1. I connected the adapter to power and the ebus connection between the heater and the Cube S Net
2. I configured the WiFi and the IP settings on the adapter with TCP and enh protocol
3. I installed the eBUSd addon in Home Assistant and configured the port and ip settings
4. I placed a file called ariston.csv in the /config folder. The file was downloaded from https://github.com/komw/ariston-bus-bridgenet-ebusd
5. I set the configuration folder as /config
 
From what I understand from the logs, the addon connects to the adapter and two devices (ebusd and ebusd eBUS device) are created in home assistant. The problem is that no entities that relate to device sensors or options are created and at the same time it seems that most of the cmd messages are unknown.

The addon configuration is the following:

scanconfig: true
loglevel_all: notice
mqtttopic: ebusd
mqttint: /etc/ebusd/mqtt-hassio.cfg
mqttjson: true
network_device: 192.168.1.95:9999
mode: enh
configpath: /config

Logs from addon:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=ohpeich6eipahngaequaijaiwithiekiephi8phahshi6aidaXo2bai4ol4eegah --scanconfig --mqttjson --configpath=/config --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:192.168.1.95:9999 --log=all:notice
2023-12-18 20:15:18.726 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.1.95:9999, enhanced
2023-12-18 20:15:18.830 [bus notice] device status: resetting
2023-12-18 20:15:18.830 [bus notice] bus started with own address 31/36
2023-12-18 20:15:18.927 [bus notice] signal acquired
2023-12-18 20:15:18.986 [bus notice] new master 37, master count 2
2023-12-18 20:15:18.986 [update notice] received unknown BC cmd: 37fe2000020120
2023-12-18 20:15:19.026 [update notice] received unknown BC cmd: 37fe200f0401200100
2023-12-18 20:15:19.184 [update notice] received unknown BC cmd: 37fe201010c04b0200091901000a1901000b190000
2023-12-18 20:15:19.223 [bus notice] new master 7f, master count 3
2023-12-18 20:15:19.223 [update notice] received unknown BC cmd: 7ffe2001026226
2023-12-18 20:15:19.242 [mqtt notice] connection established
2023-12-18 20:15:19.322 [update notice] received unknown BC cmd: 37fe200e08622664006400e001
2023-12-18 20:15:19.385 [update notice] received unknown BC cmd: 7ffe2020030b9000
2023-12-18 20:15:19.506 [update notice] received unknown BC cmd: 37fe2010100c1900000d1900000e19000001910100
2023-12-18 20:15:19.580 [update notice] received unknown BC cmd: 7ffe207309000000000000000003
2023-12-18 20:15:19.736 [update notice] received unknown BC cmd: 37fe20101002910000039100000491000005910000
2023-12-18 20:15:19.747 [bus notice] new master 70, master count 4
2023-12-18 20:15:19.777 [update notice] received unknown MS cmd: 703c2000040e11d140 / 0303010b
2023-12-18 20:15:19.824 [bus notice] device status: reset, supports info
2023-12-18 20:15:19.872 [update notice] received unknown BC cmd: 37fe20100406910000
2023-12-18 20:15:19.935 [update notice] received unknown MS cmd: 7f3c20000475470e11 / 0403160001
2023-12-18 20:15:19.937 [bus notice] device status: extra info: firmware 1.1[3a0f].1[3a0f], jumpers 0x0f
2023-12-18 20:15:20.050 [update notice] received unknown MS cmd: 3784200006d140cc4bc89c / 0407010800
2023-12-18 20:15:20.086 [update notice] received unknown BC cmd: 7ffe2001026371
2023-12-18 20:15:20.179 [update notice] received unknown BC cmd: 37fe200e086371a00064002c01
2023-12-18 20:15:20.273 [update notice] received unknown BC cmd: 7ffe2001020120
2023-12-18 20:15:20.354 [update notice] received unknown BC cmd: 37fe200e050120010001
2023-12-18 20:15:20.453 [update notice] received unknown BC cmd: 37fe2000020220
2023-12-18 20:15:20.506 [update notice] received unknown BC cmd: 37fe200f0402200100
2023-12-18 20:15:20.594 [update notice] received unknown BC cmd: 7ffe2001020220
2023-12-18 20:15:20.678 [update notice] received unknown BC cmd: 37fe200e050220010001
2023-12-18 20:15:20.775 [update notice] received unknown BC cmd: 7ffe2001020f23
2023-12-18 20:15:20.850 [update notice] received unknown BC cmd: 37fe200e050f23000001
2023-12-18 20:15:20.950 [update notice] received unknown BC cmd: 7ffe2000027647
2023-12-18 20:15:21.004 [bus notice] max. symbols per second: 144
2023-12-18 20:15:21.045 [update notice] received unknown BC cmd: 37fe200f057647370000
2023-12-18 20:15:21.130 [update notice] received unknown BC cmd: 7ffe2001026471
2023-12-18 20:15:21.229 [update notice] received unknown BC cmd: 37fe200e08647174ff74ff8c00
2023-12-18 20:15:21.407 [update notice] received unknown BC cmd: 7ffe2001026271

Koky05, 2023-12-20:
I think you have eBus v5 so device should be configured same way as is shown in eBus Adapter:
`network_device: ens:192.168.1.95:9999`

Your csv file wasn't loaded at all. But for sure you need to connect into Addon (Docker Container) (SSH into HAOS and run command `docker exec -it eBUSd /bin/bash`) and run command `ebusctl i`

john30, 2023-12-23:
check if the desired files are really available in /config to the ebusd process

mihaimdinca, 2023-12-23:
Thank you for the interest in my issue john30 and koky05. I tried what koky05 suggested and it completed successfully (apparently) after I used the addon ID in the docker exec command. Unfortunately nothing changed. As for what john30 suggested, I am not sure about how to check that, I will try to do a further reading into the documentation and if I cannot understand how to, I will try asking you for help again. Either way, I intent to post the issue or the solution here. Thank you for your support and patience. Have a wonderful day!

mihaimdinca, 2023-12-24:
So, I managed to get it working. Apparently, the problem was that the config folder was not the right one. I was using the Filebrowser addon to move files around and the config folder made available was local to the addon I believe. Anyway, I used WinSCP to access the HAOS files and realized my mistake. I moved the ariston.csv file to the right /config folder and it works now. Mostly. Thank you for your help.

closed 2023-12-24 by mihaimdinca