john30/ebusd-esp32 #43: Ebus to MQTT

vlak6613, 2023-12-30:
Hello people, I am having problems connecting the ebus adapter to Hassio.
I have been running the home assistant server on a raspberry pi and the web interface with the ebus adapter and the correct firmware is also up and running.
Only I have no idea how to get MQTT working now. I have been looking up the necessary information for a few days now, but I have no idea how to approach this now. Do I have to do something in the web interface of the ebus adapter or do I have to get started in hassio. anyone have any ideas?

vlak6613, 2023-12-30:
I managed it with the ebusd app in hassio. 

Only now I get this error. Is that because I haven't connected the boiler yet?
2023-12-30 14:33:35.436 [bus error] unable to open 192.168.178.67:9999: ERR: generic I/O error
2023-12-30 14:33:35.436 [bus notice] device invalid

Just to be clear, I have not yet connected my central heating boiler to the eBUS

john30, 2023-12-31:
is 192.168.178.67 the ip address of the adapter?
for MQTT you need to configure ebusd accordingly.

vlak6613, 2023-12-31:
Yes. Thats the correct ip.

john30, 2024-01-06:
as stated before: for MQTT you need to configure ebusd accordingly, see here: https://github.com/john30/ebusd/wiki/2.-Run#mqtt-options
not an adapter issue anyway, so closing

closed 2024-01-06 by john30