john30/ebusd-esp32 #44: Securing access to ebus when on WiFi

dulek, 2024-01-09:
Hi,

I'm trying to figure out if it's possible to have some basic authentication configured, so that I can prevent random people on my network from being able to connect to the adapter with ebusd and potentially mess with my heating device. Can I do that or do I need to run a separate tight WiFi network just for my ebus adapter?

Thanks!

Koky05, 2024-01-10:
In configuration you have access control where you can specify login and password.
[image]

dulek, 2024-01-10:
> In configuration you have access control where you can specify login and password.

That's securing access to the adapter configuration, but not the ebus stream accessible by an ebusd instance.

Koky05, 2024-01-11:
I see, you need to create separate VPN for your eBUS and eBUS Deamon or use RPi and use USB direct connection (disable WiFi) then you can setup access level on eBUS Deamon layer.

dulek, 2024-01-11:
> I see, you need to create separate VPN for your eBUS and eBUS Deamon or use RPi and use USB direct connection (disable WiFi) then you can setup access level on eBUS Deamon layer.

Okay, that's what I thought, unfortunately. Thank you for confirmation!

closed 2025-04-29 by john30