john30/ebusd-esp32 #46: adapter v5 no signal via USB from Vaillant unless initialized through web UI first

Solarer, 2024-01-20:
Hello,
this is a follow up for
#32 and https://github.com/john30/ebusd/issues/1117

I own 2 adapter v5, both connected via USB to a raspberry pi 3B running raspbian. I tried connecting them to my Vaillant Recovair 260/4 and my Vaillant Auro Compact but I always get a `no signal`. Both run the latest firmware (2024-01-06) and I tried the latest host ebusd and docker version.

In Issue #32 it was implied that the ebus signal can be aquired after initializing the ebus adapter via the web interface and selecting USB input there. I tried these steps and was able to aquire a signal from both Vaillant devices!

I am getting the feeling that this is how it is supposed to work but honestly, I was expecting ebusd to support a direct USB connection out of the box. (Especially since /dev/ttyACM0 shows up right away). If this is the intended behavior, can you add the info that only 1 connection type is supported at a time and this needs to be configured first?

Log output from a freshly flashed (erased) adapter:
```
sudo ebusd -f --scanconfig=full --device=ens:/dev/ttyACM0 --lograwdata=bytes --log=all:debug --log=network:info
2024-01-20 15:34:17.897 [main info] configPath URL is valid
2024-01-20 15:34:17.899 [bus notice] device status: transport opened
2024-01-20 15:34:17.900 [main info] registering data handlers
2024-01-20 15:34:17.900 [main info] registered data handlers
2024-01-20 15:34:17.900 [main notice] ebusd 23.3.23.3 started with full scan on device: /dev/ttyACM0, serial high speed, enhanced
2024-01-20 15:34:17.900 [main info] loading configuration files from https://cfg.ebusd.eu/
2024-01-20 15:34:18.015 [main info] reading templates /
2024-01-20 15:34:18.131 [main info] read templates in /
2024-01-20 15:34:18.131 [main info] reading file broadcast.csv
2024-01-20 15:34:18.249 [main info] successfully read file broadcast.csv
2024-01-20 15:34:18.249 [main info] reading file memory.csv
2024-01-20 15:34:18.366 [main info] successfully read file memory.csv
2024-01-20 15:34:18.366 [main info] read config files, got 11 messages
2024-01-20 15:34:18.367 [bus notice] bus started with own address 31/36
2024-01-20 15:34:18.367 [bus notice] <73
2024-01-20 15:34:18.367 [bus notice] signal acquired
2024-01-20 15:34:18.367 [bus notice] <65
2024-01-20 15:34:18.368 [bus notice] <72
2024-01-20 15:34:18.368 [bus notice] <76
2024-01-20 15:34:18.368 [bus notice] <65
2024-01-20 15:34:18.368 [bus notice] <72
2024-01-20 15:34:18.368 [bus notice] <20
2024-01-20 15:34:18.368 [bus notice] <73
2024-01-20 15:34:18.368 [bus notice] <74
2024-01-20 15:34:18.368 [bus notice] <61
2024-01-20 15:34:18.368 [bus notice] <72
2024-01-20 15:34:18.368 [bus notice] <74
2024-01-20 15:34:18.368 [bus notice] <65
2024-01-20 15:34:18.368 [bus notice] <64
2024-01-20 15:34:18.368 [bus notice] <0d
2024-01-20 15:34:18.368 [bus notice] <0a
2024-01-20 15:34:18.368 [bus notice] <65
2024-01-20 15:34:18.368 [bus notice] <61
2024-01-20 15:34:18.368 [bus notice] <73
2024-01-20 15:34:18.368 [bus notice] <69
2024-01-20 15:34:18.368 [bus notice] <3e
2024-01-20 15:34:18.368 [bus notice] <20
2024-01-20 15:34:18.368 [bus notice] <0d
2024-01-20 15:34:18.368 [bus notice] <0a
2024-01-20 15:34:18.368 [bus notice] <65
2024-01-20 15:34:18.368 [bus notice] <61
2024-01-20 15:34:18.368 [bus notice] <73
2024-01-20 15:34:18.368 [bus notice] <69
2024-01-20 15:34:18.368 [bus notice] <3e
2024-01-20 15:34:18.368 [bus notice] <20
2024-01-20 15:34:18.368 [bus notice] <0d
2024-01-20 15:34:18.368 [bus notice] <0a
2024-01-20 15:34:18.369 [bus notice] <65
2024-01-20 15:34:18.369 [bus notice] <61
2024-01-20 15:34:18.369 [bus notice] <73
2024-01-20 15:34:18.369 [bus notice] <69
2024-01-20 15:34:18.369 [bus notice] <3e
2024-01-20 15:34:18.369 [bus notice] <20
2024-01-20 15:34:18.369 [bus notice] <0d
2024-01-20 15:34:18.369 [bus notice] <0a
2024-01-20 15:34:18.369 [bus notice] <65
2024-01-20 15:34:18.369 [bus notice] <61
2024-01-20 15:34:18.369 [bus notice] <73
2024-01-20 15:34:18.369 [bus notice] <69
2024-01-20 15:34:18.369 [bus notice] <3e
2024-01-20 15:34:18.369 [bus notice] device status: reset, supports info
2024-01-20 15:34:20.029 [bus debug] ERR: read timeout during skip, switching to no signal
2024-01-20 15:34:20.029 [bus error] signal lost
2024-01-20 15:34:28.369 [main debug] performing regular tasks
2024-01-20 15:34:38.369 [main debug] performing regular tasks
```

ebusctl:
```
localhost: info
version: ebusd 23.3.23.3
device: /dev/ttyACM0, serial high speed, enhanced
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
 
After selecting USB from the web interface and rebooting:

```
sudo ebusd -f --scanconfig=full --device=ens:/dev/ttyACM0 --lograwdata=bytes --log=all:debug --log=network:info
2024-01-20 15:37:29.354 [main info] configPath URL is valid
2024-01-20 15:37:29.355 [bus notice] device status: transport opened
2024-01-20 15:37:29.355 [main info] registering data handlers
2024-01-20 15:37:29.355 [main info] registered data handlers
2024-01-20 15:37:29.355 [main notice] ebusd 23.3.23.3 started with full scan on device: /dev/ttyACM0, serial high speed, enhanced
2024-01-20 15:37:29.356 [main info] loading configuration files from https://cfg.ebusd.eu/
2024-01-20 15:37:29.470 [main info] reading templates /
2024-01-20 15:37:29.588 [main info] read templates in /
2024-01-20 15:37:29.588 [main info] reading file broadcast.csv
2024-01-20 15:37:29.706 [main info] successfully read file broadcast.csv
2024-01-20 15:37:29.706 [main info] reading file memory.csv
2024-01-20 15:37:29.823 [main info] successfully read file memory.csv
2024-01-20 15:37:29.823 [main info] read config files, got 11 messages
2024-01-20 15:37:29.824 [bus notice] bus started with own address 31/36
2024-01-20 15:37:29.824 [bus notice] <aa
2024-01-20 15:37:29.824 [bus notice] signal acquired
2024-01-20 15:37:29.824 [bus notice] <aa
2024-01-20 15:37:29.824 [bus notice] <aa
2024-01-20 15:37:29.824 [bus notice] <aa
2024-01-20 15:37:29.824 [bus notice] <aa
2024-01-20 15:37:29.824 [bus notice] <aa
2024-01-20 15:37:29.824 [bus notice] <aa
2024-01-20 15:37:29.824 [bus notice] <aa
2024-01-20 15:37:29.825 [bus notice] <aa
2024-01-20 15:37:29.825 [bus notice] <aa
2024-01-20 15:37:29.825 [bus notice] <aa
2024-01-20 15:37:29.865 [bus notice] <aa
2024-01-20 15:37:29.908 [bus notice] <aa
2024-01-20 15:37:29.950 [bus notice] <aa
2024-01-20 15:37:29.993 [bus notice] <aa
2024-01-20 15:37:30.003 [bus notice] device status: reset, supports info
2024-01-20 15:37:30.036 [bus notice] <aa
2024-01-20 15:37:30.080 [bus notice] <aa
2024-01-20 15:37:30.103 [bus notice] device status: extra info: firmware 1.1[4106].1[4106], jumpers 0x03
2024-01-20 15:37:30.122 [bus notice] <aa
2024-01-20 15:37:30.165 [bus notice] <aa
2024-01-20 15:37:30.207 [bus notice] <aa
2024-01-20 15:37:30.250 [bus notice] <aa
2024-01-20 15:37:30.293 [bus notice] <aa
2024-01-20 15:37:30.335 [bus notice] <aa
2024-01-20 15:37:30.379 [bus notice] <aa
2024-01-20 15:37:30.422 [bus notice] <aa
2024-01-20 15:37:30.464 [bus notice] <aa
```

```
ebusctl
localhost: info
version: ebusd 23.3.23.3
device: /dev/ttyACM0, serial high speed, enhanced, firmware 1.1[4106].1[4106]
signal: acquired
symbol rate: 8
max symbol rate: 29
min arbitration micros: 5
max arbitration micros: 23
min symbol latency: 4
max symbol latency: 5
scan: finished, some messages pending
reconnects: 0
masters: 2
messages: 12
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=recoV;SW=0229;HW=0203"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
```

john30, 2024-01-20:
this is documented in several locations like the [first steps](https://adapter.ebusd.eu/v5/steps.en.html#adaptercfg) and also the [main page](https://adapter.ebusd.eu/v5/index.en.html#variants) in every section like "By using the easy> interface the connection is set to...". if I add it in further locations (that are obviously only rarely read), it would blow it up even more (leading to even more people not reading it).

nevertheless, if you have an idea how to make it more obvious, let me know.

closed 2024-01-20 by john30

Solarer, 2024-01-20:
ok, maybe `By using the [easi> interface], the host connection is set to USB` could be a little rephrased because I did read the USB section but did not understand what this meant.
Maybe like this: `Before ebusd can be connected to the adapter, the host connection must be changed from TCP to USB. This can be done via easi> interface or via web GUI.`

Also, an error message from ebusd if someone tries the wrong host connection would be really helpful. Because in https://github.com/john30/ebusd/issues/1117 even You could tell from the log output what the issue was.

john30, 2024-01-20:
that text is actually a link explaining what it is, so again I wouldn't know how to make that more explicit without blowing it up unnecessarily.

how would ebusd know where your adapter sits?? and there is an error message in the log anyway in case the device connection is invalid: "device invalid"

Solarer, 2024-01-20:
I only got the `device invalid` message when I used a wrong path (e.g. /dev/ttyACM1 instead of /dev/ttyACM0) or wrong IP/port or if the device was blocked by another ebusd instance. But not when the host connection was TCP and I attempted a connection via USB. In that case /dev/ttyACM0 is present and the device not invalid. I did not receive an error message then.