john30/ebusd-esp32 #47: eBUS Adapter Shield v5 & Raspberry Pi 5

Thde23, 2024-01-29:
Hello,

I have already got the eBUS Adapter v5 running completely under the following configuration and have generated data points under ioBroker.
- eBUS Adapter v5 mounted on GPIO strip of a Raspberry Pi 3b
- Raspberry Pi OS (Legacy - Debian 11) 

Now I have plugged eBUS Adapter v5 onto a Raspberry Pi 5 (Raspberry Pi OS (Debian 12)). 
Then set everything up as described in the instructions at https://github.com/john30.
Unfortunately I get no signal at `ebusctl info`.

When restarting (power), the blue LED only flashes once for 0.5s. The green and red LEDs do not light up at all!
When the button is pressed, only the blue LED lights up (voltage present).
I have activated and set up both Wifi Station/Clien and WiFi Access Point in the interface ("easi>"). 
A connection is established if the eBUS adapter is mounted on the Raspberry Pi 3b or is only operated separately with a 5V power supply unit. 
If the adapter is mounted on the Raspberry Pi 5, a connection can no longer be established. 

I suspect that the eBUS Adapter v5 is not booting correctly when it is mounted on the Raspberry Pi 5. 
I used a Siemens 5V/3A control cabinet power supply.  

`ls -l /dev`
[image]
`dmesg | grep ttyAMA`
[image]
`pinctrl get 0-15`
[image]

I would be grateful for any suggestions and hope the translation from german to english is reasonably understandable.

john30, 2024-02-04:
I tested it recently on a RPi 5 and it worked flawlessly. can you check if it works when only connecting the 4 first pins on the W2 connector?
ah, the blue led flashes during boot? thats a problem as it will put the ESP32-C3 into download mode. so you have something configured on the Raspi that pulls IO9 on the adapter / PIN31 on the Raspi low during boot and that needs to be removed (maybe some process on the Raspi or a setting in the /boot/config.txt)

Thde23, 2024-02-06:
Many thanks for the support!

I have successfully connected the adapter via the first 4 pins on the W2 and received data from the heating control with ebusd.
[image]
[image]

Here is a printout of the /boot/config.txt.
[image]

The status of the interface ("easi>") now also shows the connection to the eBus and ebusd. 
I don't understand why the ebusd is displayed as (inactive)? 

[image]

Unfortunately, I have to revise my information on the LED status when restarting (power) and the eBUS adapter is connected normally:
All three (blue, red, green) LEDs flash once for approx. 200ms. Then nothing more. I don't know whether I didn't see it correctly or whether something has changed.

I only had to make an adjustment in the ebusd config (/etc/default/ebusd).
`EBUSD_OPTS=" --device=ens:/dev/ttyAMA10 ...`  ->  `EBUSD_OPTS=" --device=ens:/dev/ttyAMA0 ...`
I don't understand that either, as serial0 is assigned to ttyAMA10 (see https://github.com/john30/ebusd-esp32/issues/47#issue-2105881154)?

> so you have something configured on the Raspi that pulls IO9 on the adapter / PIN31 on the Raspi low during boot
>
The eBUS adapter does not occupy PIN31. Do you mean PIN13 (GPIO27) of the RPi 5?
[image]

john30, 2024-02-11:
I guess with enable_uart=1 it won't work as the port is then used by a system console (this is only needed on older OS versions), so you probably answered yes to the login shell over serial here https://adapter.ebusd.eu/v5/raspberrypi.en.html#configuration-with-raspberry-pi-os
yes, I meant pin 13. could be 18 as well

stroel, 2024-03-12:
Hi @Thde23 

It looks like I have the same problem as you.
I also have the v5 Adapter Shield with a Raspberry Pi 5.
When I mount the shield onto my Raspberry Pi, using a stack header, it isn't working.

I have changed the settings of the Adapter Shield so it's working through WiFi and now I see all kinds of data coming in through MQTT. 

[image]

So the Adapter Shield is working as expected, but somehow I can't connect it through my Raspberry Pi.

Before I try to use the only first 4 pins. Is it already working as it should?
I also had the /dev/ttyAMA10 instead of the /dev/ttyAMA0 and you had to define the latter instead of the first one.

Thanks!

Thde23, 2024-03-12:
Hello @stroel,

at the weekend I identified pin 13 (GPIO 27) as the cause of the problem.
I have now shortened two 10x1 stack headers accordingly and cut off pin 13. 
That's how it works;-)  
[image]

I have not yet been able to find out exactly what is wrong with pin 13. 
I have just started with Raspberry Pi and Linux. So my knowledge is still very limited and I can't find the right approach or solution at the moment.

stroel, 2024-03-12:
Hi @Thde23 

Thank you for your feedback.
I can confirm this is working!

But one thing which is still strange, like you said before, I see that I have /dev/ttyAMA10 instead of /dev/ttyAMA0.
For some odd reason I still have to configure the AMA0 instead of the AMA10 in the ebusd config.

Anyway, I see data coming in through MQTT.
Thanks again!

john30, 2024-03-24:
I've tested it with Raspi 3B, 4B, and 5, measured with my osci, and found out that the timing and level during power up are quite different on the 5, so this can only work when removing the pin in the connection. [instructions are placed to the docs](https://adapter.ebusd.eu/v5/raspberrypi.en.html#rpi5) and future versions will no longer have the issue with Raspi 5

closed 2024-03-24 by john30

stroel, 2024-03-24:
Thanks @john30 

Still working perfectly. Thanks!