john30/ebusd-esp32 #49: Firmware on generic ESP32-C3 with eBUS Adapter 2

LeZerb, 2024-02-16:
Currently I have an eBUS Adapter 2 connected to a Pi Zero.

Would it be possible to use the eBUS Adapter Shield v5 firmware on a generic ESP32-C3 (e.g. ESP32-C3 SuperMini from eBay) when connecting I/O 7 and 10 to an eBus Adapter 2?

I already have an adapter shield v5 on a seperate setup which seems to behave better than the PI+Adapter2 setup.

LeZerb, 2024-02-19:
I started to look into it.

When I only connect Rx I see some raw messages coming in but I am not sure if these are correct.
When I connect Tx of the ESP32 C3 board the Tx LED on Adapter 2 is always on. The Tx input on adapter 2 seems to be low active. And from what I can see the ESP32 firmware does not toggle I/O 10 (ebus Tx) at all it always stays low.

I would expect to see some activity on I/O 10. Maybe someone can point me in the right direction?

LeZerb, 2024-02-19:
One additional note.

During initialization I am getting this information.

device status: extra info: firmware 1.1[3c17].1[3c17], jumpers 0x0f

Could this issue be related to the jumper states? Maybe one of those means something like a "monitor mode". Is there information available what these bits mean exactly?

LeZerb, 2024-02-24:
I have been able to get it to work by inverting the signals between the ESP32C3 board and the Ebus Adapter v2. At least I am now getting "eBUS signal: acquired". 

During initialization I am getting the information "setup_uart: eBUS UART 1 on TX=10(inv) RX=7(inv) with 2400 Bd"
Is it possible to initialize the UART with IO7 and IO10 _not_ inverted? I have tried setting pin.7.func and pin.10.func to "i" (I thought this might mean inverted) which did not change the behaviour.

If that is not possible yet would it be possible to add a way to invert the UART behaviour maybe by setting a specific NVS value or with a "hidden" switch to the ebus easi command?

LeZerb, 2024-02-27:
I have found a solution. For a more detailed description you can have a look at [the discussion in the eBus repo](https://github.com/john30/ebusd/discussions/978) (in german)

LeZerb, 2024-02-27:
It would however still be nice to have the possibility to configure the Rx and Tx pins in a more user friendly way.

john30, 2024-05-26:
I've added a release targeting the Wemos C3: https://adapter.ebusd.eu/v5/wemos.en
this can be used for the purpose as well

closed 2024-05-26 by john30

Mozzarella7, 2024-12-01:
Hi @john30, 
could you share ebus protocol (enc/plain) and pin setup for WEMOS C3 and adapter V2.x. I am trying to replace old wemos D1 with C3 please. Thanks
Rafal