john30/ebusd-esp32 #50: allow 64 char wifi passsword

john30, 2024-02-27:
moved from https://github.com/john30/ebusd/issues/1201

BastelBaus, 2024-03-03:
Thanks John for moving it here.
Any idea when you could fix it or any chance that I can help fixing it or getting the sources to fix it locally for me ?
BR,
BastelBaus

john30, 2024-03-09:
you can get the latest version currently being tested by adding "?file=testing.bin" to the adapter host name/ip address of your browser and then flash that version with the "update firmware" button (an additional entry will appear there)

BastelBaus, 2024-03-09:
Thank you for this quick fix and the guidance how to apply! Works like charm!

Just as side issue (no problem for me since I just used another network to flash), I was not able to upload the firmware while I was connected to the ebusd-esp as WiFi AP, Got the following error message:
[image]

john30, 2024-03-10:
of course it can't work in AP mode - how would it be able to download the file from internet?

anyway, closing this as it is solved with the new version released today

closed 2024-03-10 by john30