john30/ebusd-esp32 #54: Wifi Error CCMP replay detected

JosemyAB, 2024-03-31:
I'm using the v5 adapter with this setup:
[image]

**Ebus mode is plain**

I'm runing the ebusd deamon in a docker container with this setup:
```
# EBUSD
  ebusd: 
    container_name: ebusd 
    image: john30/ebusd:v23.3
    privileged: true
    environment: 
      EBUSD_DEVICE: "192.168.1.170:9999" 
      EBUSD_ACCESSLEVEL: "*" 
      EBUSD_SCANCONFIG:
      EBUSD_CONFIGPATH: "/ebusd-configuration/latest/en" 
      EBUSD_POLLINTERVAL: 10
      EBUSD_LATENCY: 20
      EBUSD_MQTTINT: "/ebusd-configuration/mqtt-hassio.cfg"
      EBUSD_MQTTJSON: ""
      EBUSD_MQTTHOST: "192.168.1.6" 
      EBUSD_MQTTPORT: "1883"
      EBUSD_MQTTUSER: "****"
      EBUSD_MQTTPASS: "****"
      TZ: "Europe/Madrid"      
    volumes: 
      - /docker/ebusd/etc:/etc/ebusd 
      - /docker/ebusd/ebusd-configuration:/ebusd-configuration
      - /docker/ebusd/var:/var
      - /etc/localtime:/etc/localtime:ro 
    restart: always 
    depends_on: 
      - mosquitto
```

And I get a lot of warnings:
```
Warning	
host: errors: protocol```

And this error:
```
wifi:
19:38:49		
CCMP replay detected: A1=dc:54:75:92:8c:9c A2=04:d9:f5:c5:f5:30 PN=21749, RSC=21750 seq=52960
```
<img width="982" alt="image" src="https://github.com/john30/ebusd-esp32/assets/13203575/843c95a6-ff0d-4952-8340-3361270b579b">

I'm running the last version of the firwmare (2024-03-30) and ebusd 23.3

Am I doing something wrong?

JosemyAB, 2024-03-31:
I attach the log file of the Ebusd deamon. I don't see anything related, but just in case
[_ebusd_logs.txt](https://github.com/john30/ebusd-esp32/files/14817123/_ebusd_logs.txt)

john30, 2024-04-01:
why do you use plain mode? thats basically dropping all the adantages in the first place.
never saw CCMP messages before, so might be something with your access point and you should probably check there

JosemyAB, 2024-04-01:
With the enhanced mode I have some typical error messages like "arbitration lost", "ERR: SYN received", etc. With the plain mode I don't get any of them. In any case these errors are not serious at all, are they?

On the other hand, what about the "host: errors: protocol" warnings? What do they mean?

john30, 2024-04-20:
a bit of googling reveals that this replay could really mean that somebody is trying to hack you.
anyway, you might want to try the [newest version](https://adapter.ebusd.eu/v5/ChangeLog#20240420) just released as there were tons of commits related to wifi in ESP-IDF again

john30, 2024-05-05:
please check with the [new version 20240505](https://adapter.ebusd.eu/v5/ChangeLog#20240505) just published if this is still the case

JosemyAB, 2024-09-05:
Ey John! sorry for the late response! 
Today I've updated the adapter to the last version [20240825](https://adapter.ebusd.eu/v5/ChangeLog#20240825)

Now I don't have the CCMP error, so it has been fixed. On the other hand the enhanced mode is working well now for my machine

Thanks for your hard work!

I have a couple new things. They don't look very bad but just in case you think they should be fixed:
* read_host_tcp: recv: errno 104
* read_host_tcp: recv: errno 128

This is the log:
[image]

john30, 2024-10-08:
these happen when the host (ebusd here) closes the connection, so nothing unusual but logged anyway to be aware of potential other socket errors that might occur

closed 2024-10-08 by john30