john30/ebusd-esp32 #55: default device string change

renco33, 2024-04-17:
Begginer question. How to change default device string? Configured adapter with W5500 to be on 192.168.0.170:9999 and its web iterface works. wifi is off.
Also modified  /etc/default/ebusd  to have following line:
EBUSD_OPTS="scanconfig -d ens:192.168.0.170:9999"
all other lines commented out with "# "

Bust ebusctl still searches it on /dev/ttyUSB0:

user@raspberrypi:~ $ ebusctl i
version: ebusd 23.3.23.3
update check: OK
device: /dev/ttyUSB0, serial, invalid
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

What is wrong or missing here?

renco33, 2024-04-17:
figured out: sudo systemctl enable ebusd  and reboot did the trick. However other command "sudo chkconfig ebusd on", suggested at https://github.com/john30/ebusd/wiki/2.-Run   does not work - chkconfig: command not found...

john30, 2024-04-27:
not related to the firmware, but anyway. it is written just one line above in the wiki:
`sudo systemctl enable ebusd`  
`sudo chkconfig ebusd on`

closed 2024-04-27 by john30