john30/ebusd-esp32 #56: After Upgrade to 2024-04-15 "API retrieval failed for /status: timeout"

kimzeuner, 2024-04-18:
After the latest update i get the “ API retrieval failed for /status: timeout“ error (see also screenshot). The webinterface is still available but it seems as my ethernet module USR-ES1 is not detected correctly. 
[image]

Seems to be a similar problem as in #38 

how can i downgrade to the version before ?

gasman1844, 2024-04-19:
Upgrading to 2024-04-15 I have seen the same error message but I am using the wifi connection, not ethernet. No ebus signal acquired. Tried rebooting adapter and also reset daemon but didn't work. Went back to 2004-03-30 and everything OK

john30, 2024-04-20:
without any logging is hard to tell and I can't reproduce with any of the two USR-ES1 I have.
so please connect via USB to a host and use the "Logs & Console" option of the firmware page and then use "log -l 0" to get all available logs.
you might also want to try with the [newest version](https://adapter.ebusd.eu/v5/ChangeLog#20240420) just released

kimzeuner, 2024-04-20:
Thanks for your feedback. Just updated to 2024-04-20 and everything seems to be working fine. 
Are the logs from 2024-04-15 still interesting for you for further investigation? I could go back to 2024-04-15 and share the logs if needed. Otherwise i will close the issue?!

closed 2024-04-20 by john30