john30/ebusd-esp32 #58: easi> Configuration page, MQTT Broker URI can't handle more than 32 character instead of the reported 64

etmmafl, 2024-04-22:
The problem is in easi> Configuration page, **MQTT Broker URI** 
The text field can't handle more than 32 character instead of the reported 64

I had to truncate my login and password details to get this working.

[image]

Build: [20240420]
WiFi station/client: XXXXXXX 192.168.68.145, 90% (-55dBm)
WiFi access point: inactive
Chip ID: 348518f7c6d8, [ESP32-C3, rev 4]
Hostname: ebus-xxxxxxx
ebusd connected: no
eBUS signal: no signal
MQTT connected: yes

john30, 2024-04-27:
good spot, thanks. just a little too tight validation here :) will be fixed in next release.
you can circumvent for now by using the repl, i.e. `nvs_set mqtt.uri str -v "..."`

Curtain4win, 2024-05-02:
> good spot, thanks. just a little too tight validation here :) will be fixed in next release. you can circumvent for now by using the repl, i.e. `nvs_set mqtt.uri str -v "..."`

unfortunately then it does not allow to reboot or save via gui of easi>

john30, 2024-05-05:
it is saved when using that command and a reboot is as simple as entering "restart" to the repl.
anyway, fixed with the [new version 20240505](https://adapter.ebusd.eu/v5/ChangeLog#20240505) just published

closed 2024-05-05 by john30