john30/ebusd-esp32 #59: adapter stopped working with "bus: errors: overrun: messages.

cnu80, 2024-05-02:
Hi, with the newest firmware version (Version 2024-04-20) my adapter is working for some hours, than I got many overrun errors   and I have to reboot or switch off/on the adapter.

[image]

Any ideas ? thanks

JavanXD, 2024-05-03:
I do have the same issue, I switched from wifi to usb, then from usb to ethernet but still, after some time those "overrun" error messages appear both in ebusd HA-Add-On log and in ebus Adapter log. Rebooting the adapter works for me.
[image]

I wrote myself a workaround, which is tracking the ebus Signal status in Homeassistant, and then reboots the adapter if it keeps stuck again. 

```yaml
sensor:
- platform: rest
  name: "eBUS Adapter JSON" 
  resource: http://192.168.178.XX/api/v1/status
  json_attributes_path: "ebus"
  value_template: "{{ value_json.ebus.signal }}"
  json_attributes:
    - on
    - connected
    - proto
    - host
    - signal
  headers:
    Content-Type: application/json
  scan_interval: 60

rest_command:
  ebus_reboot:
    url: "http://192.168.178.XX/api/v1/config?reboot=1"
    method: put
    timeout: 2
    content_type: "application/json"
    payload: '{"httpd":{"tls":false,"user":"","pass":""},"sta":{"on":true,"ssid":"XXX","pass":"XXX"},"ap":{"on":false},"ebus":{"plain":false,"ping":true,"ledRxTx":true,"proto":"ens","host":"tcp"},"pin":{"4":{"func":""},"7":{"func":""},"9":{"func":""},"10":{"func":""},"20":{"func":"lr"},"21":{"func":"lt"},"0":{"func":"ei"},"1":{"func":"ohf"},"2":{"func":"eo"},"3":{"func":"ec"},"5":{"func":"et"},"6":{"func":"es"},"8":{"func":"iuf"}},"eth":{"on":true},"hostname":"vaillant-ebus","mqtt":{"ver":0,"uri":"mqtt://XX:XX@192.168.178.XX"},"caps":{"mqtts":true}}'
```
    
Automation:
```yaml
alias: eBUS Adapter Reboot wenn Signal inaktiv
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.ebus_adapter_json
    attribute: signal
    for:
      hours: 0
      minutes: 10
      seconds: 0
    to: "False"
condition: []
action:
  - service: rest_command.ebus_reboot
    data: {}
    continue_on_error: true
  - delay:
      hours: 0
      minutes: 10
      seconds: 0
      milliseconds: 0
mode: single
```

cnu80, 2024-05-04:
thanks for your hass workaround !!!.  I used a curl command, but your solution with json is better.

john30, 2024-05-05:
please check with the [new version 20240505](https://adapter.ebusd.eu/v5/ChangeLog#20240505) just published if this is still the case

cnu80, 2024-05-08:
New fw is running for two days and the overrun errors are gone.  Now I have "host errors", but the adapter is working, and I get the data from my Vaillant heat pump faster.

john30, 2024-05-12:
thanks for checking

closed 2024-05-12 by john30