john30/ebusd-esp32 #61: Wrong ebusd device string if ethernet is enabled

JavanXD, 2024-05-03:
Problem:
ebusd Adapter Info page shows the wrong ebusd device string if ethernet is enabled. The IP in the device string is wrong one. 

Screenshot taken with AP enabled:
[image]
But also, if WiFi is enabled and Ethernet active, the device string IP is still the one for WiFi - but every normal user would use the Ethernet IP instead.

JavanXD, 2024-05-03:
One more question to the shown device string:
I assumed `enh` stands for the enhanced protocol. But the device string in the info banner shows the `ens:` prefix.
I observed in some GitHub issues that a couple of people are using the `ens:` prefix even if the enhanced protocol is enabled.

john30, 2024-05-05:
done with the [new version 20240505](https://adapter.ebusd.eu/v5/ChangeLog#20240505)