john30/ebusd-esp32 #63: [bug] dangling eBus signal while WiFi enabled

JavanXD, 2024-05-03:
With my V5 Shield Adapter I had huge difficulties and I didn't know where they came from. I tried wifi, usb, and Ethernet but I still had lot's of overrun errors, on all recent release versions. 

All the time I had the WiFi enabled, not for the data transfer but for being able to access the webinterface. 

Today I found out, that disabling the WiFi on the esp32 fixes the disconnecing ebus signal. 

I wrote a small rest sensor in Home Assistant which monitores the ebus signal status.

After I turned off the WiFi in the Adapter Configuration, the random ebus signal loses went away.
```
WiFi station/client: inactive
WiFi access point: inactive
```
And I am still able to access the webinterface via the IP address from the ethernet plug. 

[image]

In the screenshot you can see the purpe little lines which showed that the ebus Signal was disconnected. After I disabled the WiFi at 1pm the signal kept constantly working.

john30, 2024-05-05:
please check with the [new version 20240505](https://adapter.ebusd.eu/v5/ChangeLog#20240505) just published if this is still the case

JavanXD, 2024-05-05:
Hi @john30 , also with the new version the ebus Signal is very soon lost and the "overrun" error messages appear, see screenshot below:
[image]
Once I disable the WiFi the Adapter works again.

john30, 2024-05-06:
what is the signal strength according to the status page?

JavanXD, 2024-05-07:
Before the firmware update I was able to connect in parallel to ethernet and to wifi.
After the update - while being connected to ethernet, the ebus Adaper cannot connect to my Wifi network, the Wifi status stays "not connected", while it creates the following error messages: 
[image]
Looks like it was not able to connect to my WiFi - the router is only 30cm away from the ebus Adapter. 

But what works is running the WiFi Access Point on the Adaper while being connected via Ethernet.

JavanXD, 2024-05-07:
Note the fact that the wifi commands are causing the "overrun" errors. 
But I am not using Wifi, I am connecting the ebusd container via the ethernet-ip-address.

john30, 2024-05-12:
if you're so close to the router, then maybe lowering the tx power might help (can be done with recent firmware in the web ui). is there any logging in the router?

john30, 2025-04-29:
closed for inactivity

closed 2025-04-29 by john30