john30/ebusd-esp32 #64: Ebus Adapter Crashes with 2024-04-20

jabdoa2, 2024-05-04:
I updated my adapter to 2024-04-20 (from 2024-03-30) and it became quite unstable with that version. WebUI breaks. I.e. the /status endpoint no longer works and I cannot reboot anymore via the webui. MQTT still published and ebus also still works.

Any idea on how to solve that?

jabdoa2, 2024-05-04:
I let it run over night (since I did not want to go down just yesterday). MQTT actually stopped working. Ebus continued. WebUI did not recover.

Rebooted it today and it started working again. I disabled MQTT for now to see if that helps.

jabdoa2, 2024-05-04:
Seems to be stable without MQTT

john30, 2024-05-05:
please check with the [new version 20240505](https://adapter.ebusd.eu/v5/ChangeLog#20240505) just published if this is still the case

jabdoa2, 2024-05-14:
Seems to be stable with the latest release. Thanks!

jabdoa2, 2024-05-14:
Just after I closed the issue it occurred again. WebUI crashed again. Guess I will disable MQTT again for now.

john30, 2024-05-18:
might be related to mqtt messages sent to the device. so if it occurs again, please check which messages were sent to the topic prefix.
anyway, there was a little issue with extracting the topic suffix, so you might want to try with the [new release](https://adapter.ebusd.eu/v5/ChangeLog#20240518)

jabdoa2, 2024-05-21:
Today it crashed with MQTT disabled. I did restart ebusd a few times which might have contributed. I took the opportunity to upgrade to 2024-05-18.

Any idea what else I should do to stabilize it? I already added a remote controlled plug adapter to automate restarts but that is more a stop gap solution for now.

john30, 2024-05-22:
so did it with the former version or the current one?
you could check the power supply and/or the USB cable as these are often an issue.
if that does not improve the situation, i would need you to be more concrete and also supply logging along with the browser being used, as well as how it is connected (tunneling/proxy or direct), other participants potentially accessing the ui etc

john30, 2025-04-29:
closed for inactivity

closed 2025-04-29 by john30