john30/ebusd-esp32 #65: ebusd does not see the device

Andrew55529, 2024-05-04:
I have a protherm ray boiler, but the connection is not working.

> 2ad9b828-ebusd:/# ebusctl find
broadcast datetime = no data stored
broadcast error = no data stored
broadcast id = no data stored
broadcast id = no data stored
broadcast signoflife = no data stored
memory eeprom = no data stored
memory ram = no data stored`

In Home Assistant Add-on: eBUSd:

> 2024-05-05 00:16:12.623 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:12.667 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:12.710 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:12.753 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:12.796 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:12.839 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:12.883 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:12.927 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:12.970 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.029 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.057 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.113 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.145 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.190 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.233 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.274 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.318 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.362 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.462 [bus debug] ERR: read timeout during receive command, switching to skip
2024-05-05 00:16:13.950 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:13.986 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.030 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.075 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.119 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.220 [bus debug] ERR: read timeout during receive command, switching to skip
2024-05-05 00:16:14.701 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.745 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.789 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.832 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.876 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.920 [bus debug] ERR: SYN received during receive command, switching to ready
2024-05-05 00:16:14.964 [bus debug] ERR: SYN received during receive command, switching to ready

john30, 2024-05-05:
these are debug messages and as such not revealing anything relevant. did you use the right device string to ebusd as shown on the adapter web page?

Andrew55529, 2024-05-05:
Yes. In adapter web page: [ebusd device string]: ens:192.168.9.138:9999
And in HA addon ebusd i write: 192.168.9.138:9999

Andrew55529, 2024-05-05:
I set log level to info and restart addon:

> s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=ooyai5seeth3phahd8o --scanconfig --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.9.138:9999 --log=all:info --scanconfig
2024-05-05 22:10:28.739 [main info] configPath URL is valid
2024-05-05 22:10:28.739 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.9.138:9999
2024-05-05 22:10:28.739 [main info] loading configuration files from https://cfg.ebusd.eu/
2024-05-05 22:10:29.059 [main info] reading templates /
2024-05-05 22:10:29.387 [main info] read templates in /
2024-05-05 22:10:29.387 [main info] reading file broadcast.csv
2024-05-05 22:10:29.750 [main info] successfully read file broadcast.csv
2024-05-05 22:10:29.750 [main info] reading file memory.csv
2024-05-05 22:10:30.068 [main info] successfully read file memory.csv
2024-05-05 22:10:30.068 [main info] read config files, got 11 messages
2024-05-05 22:10:30.100 [bus notice] bus started with own address 31/36
2024-05-05 22:10:30.100 [main info] registering data handlers
2024-05-05 22:10:30.100 [mqtt info] mosquitto version 2.0.18 (compiled with 2.0.15)
2024-05-05 22:10:30.103 [main info] registered data handlers
2024-05-05 22:10:30.105 [mqtt notice] connection established
2024-05-05 22:10:30.130 [bus notice] signal acquired

john30, 2024-05-06:
according to the ebusd log the device string is not correct (you missed the "ens:" prefix)

Andrew55529, 2024-05-06:
after add prefix its start, but i gpt new error

> 2024-05-06 20:26:13.724 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0108;7503
2024-05-06 20:26:13.724 [bus notice] scan 08: ;Vaillant;BAI00;0108;7503
2024-05-06 20:26:13.902 [network info] [00002] connection closed
2024-05-06 20:26:13.902 [network info] [00002] client connection opened 127.0.0.1
2024-05-06 20:26:14.070 [main info] reading templates vaillant
2024-05-06 20:26:14.417 [main info] read templates in vaillant
2024-05-06 20:26:30.257 [network error] HTTP connect: timed out after 10 sec
2024-05-06 20:26:30.257 [main error] HTTP failure, repeating: not connected
2024-05-06 20:26:31.258 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 20:26:31.258 [main error] error reading common config file vaillant/scan.csv: ERR: element not found, 
2024-05-06 20:26:31.258 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 20:26:31.258 [main error] HTTP failure, repeating: not connected
2024-05-06 20:26:32.258 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 20:26:32.258 [main error] error reading common config file vaillant/broadcast.csv: ERR: element not found, 
2024-05-06 20:26:32.258 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 20:26:32.258 [main error] HTTP failure, repeating: not connected
2024-05-06 20:26:33.259 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 20:26:33.259 [main error] error reading common config file vaillant/general.csv: ERR: element not found, 
2024-05-06 20:26:33.259 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 20:26:33.259 [main error] HTTP failure, repeating: not connected
2024-05-06 20:26:34.259 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 20:26:34.259 [main error] error reading scan config file vaillant/08.bai.csv for ID "bai00", SW0108, HW7503: ERR: element not found, 
2024-05-06 20:26:34.259 [main error] scan config 08: ERR: element not found
2024-05-06 20:26:34.259 [main info] loading configuration files from https://cfg.ebusd.eu/
2024-05-06 20:26:34.260 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 20:26:34.260 [main error] HTTP failure, repeating: not connected
2024-05-06 20:26:35.261 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 20:26:35.261 [main error] error reading config files from https://cfg.ebusd.eu/: ERR: element not found, last error:

Andrew55529, 2024-05-06:
2024-05-06 20:32:16.862 [main info] configPath URL is valid
2024-05-06 20:32:16.863 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.9.138:9999, enhanced
2024-05-06 20:32:16.863 [main info] loading configuration files from https://cfg.ebusd.eu/
2024-05-06 20:32:17.211 [main info] reading templates /
2024-05-06 20:32:17.559 [main info] read templates in /
2024-05-06 20:32:17.559 [main info] reading file broadcast.csv
2024-05-06 20:32:17.878 [main info] successfully read file broadcast.csv
2024-05-06 20:32:17.878 [main info] reading file memory.csv
2024-05-06 20:32:18.218 [main info] successfully read file memory.csv
2024-05-06 20:32:18.218 [main info] read config files, got 11 messages
2024-05-06 20:32:18.248 [bus notice] device status: resetting
2024-05-06 20:32:18.248 [bus notice] bus started with own address 31/36
2024-05-06 20:32:18.248 [main info] registering data handlers
2024-05-06 20:32:18.248 [mqtt info] mosquitto version 2.0.18 (compiled with 2.0.15)
2024-05-06 20:32:18.250 [bus notice] device status: reset, supports info
2024-05-06 20:32:18.251 [main info] registered data handlers
2024-05-06 20:32:18.252 [bus notice] device status: extra info: firmware 1.1[4505].1[4505], jumpers 0x0b
2024-05-06 20:32:18.270 [bus notice] signal acquired
2024-05-06 20:32:18.329 [mqtt notice] connection established
2024-05-06 20:32:27.603 [bus notice] new master 00, master count 2
2024-05-06 20:32:28.252 [bus info] scan 05 cmd: 3105070400
2024-05-06 20:32:28.872 [bus info] arbitration delay 10 - 10 micros
2024-05-06 20:32:28.880 [bus info] send/receive symbol latency 8 - 8 ms
2024-05-06 20:32:28.887 [bus info] send/receive symbol latency 6 - 8 ms
2024-05-06 20:32:28.981 [main error] scan config 05: ERR: read timeout

Andrew55529, 2024-05-06:
Now it cant find csv

> 2024-05-06 22:05:25.908 [update info] sent MS cmd: 3108070400 / 0ab5424149303001087503
2024-05-06 22:05:25.908 [bus notice] scan 08: ;Vaillant;BAI00;0108;7503
2024-05-06 22:05:25.908 [update notice] store 08 ident: done
2024-05-06 22:05:25.908 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0108;7503
2024-05-06 22:05:25.908 [bus notice] scan 08: ;Vaillant;BAI00;0108;7503
2024-05-06 22:05:26.268 [main info] reading templates vaillant
2024-05-06 22:05:26.619 [main info] read templates in vaillant
2024-05-06 22:05:27.291 [main notice] read common config file vaillant/scan.csv
2024-05-06 22:05:27.637 [main notice] read common config file vaillant/broadcast.csv
2024-05-06 22:05:27.981 [main notice] read common config file vaillant/general.csv
2024-05-06 22:05:43.543 [network error] HTTP connect: timed out after 10 sec
2024-05-06 22:05:43.543 [main error] HTTP failure, repeating: not connected
2024-05-06 22:05:44.543 [network error] HTTP new_ssl_conn: 167772567=shutdown while in init
2024-05-06 22:05:44.543 [main error] error reading scan config file vaillant/08.bai.csv for ID "bai00", SW0108, HW7503: ERR: element not found, 
2024-05-06 22:05:44.543 [main error] scan config 08: ERR: element not found
2024-05-06 22:05:55.867 [bus notice] new master 00, master count 3
2024-05-06 22:05:56.544 [bus info] scan 05 cmd: 3105070400
2024-05-06 22:05:57.135 [bus info] arbitration delay 3 - 10 micros
2024-05-06 22:05:57.149 [bus info] send/receive symbol latency 5 - 14 ms
2024-05-06 22:05:57.242 [main error] scan config 05: ERR: read timeout

Andrew55529, 2024-05-06:
Thank you. My boiler was found

address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0108;HW=7503", loaded "vaillant/bai.0010008045.inc" ([PROD='0010023648']), "vaillant/08.bai.csv"

How can I understand which data is updated automatically and which is not?

in .inc file only type r, not r[1-9]

Andrew55529, 2024-05-06:
And i have some error:

> [update info] sent MS cmd: 3108b509030dd500 / 00
[update error] unable to parse poll-read bai VortexFlowSensor from 3108b509030dd500 / 00: ERR: invalid position
[mqtt error] decode bai VortexFlowSensor: ERR: invalid position

> [update info] sent MS cmd: 3108b509030d2500 / 00
[update error] unable to parse poll-read bai externalFlowTempDesired from 3108b509030d2500 / 00: ERR: invalid position
[mqtt error] decode bai externalFlowTempDesired: ERR: invalid position

john30, 2024-05-12:
that depends on the ebusd-configuration files used and what is set to polling in there. additionally messages can be requested for polling via mqtt and so does the mqtt integration for hassio if configured accordingly. see here: https://github.com/john30/ebusd/wiki/MQTT-integration#home-assistant

closed 2024-05-12 by john30