john30/ebusd-esp32 #66: Protocol error

kenny00111, 2024-06-06:
I'm using ebusd in the following config:
[
[image]
](url)
[image]

And I'm flooded with protocol errors.
[image]

Besides this everything is working. I can control the boiler, get sensor statuses etc.
Should I ignore, or what can be the cause of it?
Thanks

worstover, 2024-06-06:
Hi Kenny00111.

I have exactly the same problem, except that it leads to the point that the ebus system is not working properly after a while. Even my Vaillant VC620 loses information about the components on the bus. So it seems to me that the adapter is disturbing the bus in any way.

A reboot of the adapter solves the problem for a while (some hours) but then the same happens again. This is why I have started a discussion about how to reboot the apdapter automatically or remotely by an URL call, but did not receive an answer.

To my mind the behaviour was different in the past and started with any firmware update. But I can't exactly say when. This is why I have also asked for a way to flash the adapter with any older firmware. unfortunately also without any response.

At the moment I'm quite lost and reboot the adapter manually whenever it comes to my mind.

MichaWilk, 2024-06-15:
Hallo,
hab genau das gleiche Problem mit der Version 2024-05-26 und Version 2024-05-18.
Kann ich auf eine ältere Version 2024-04-20 zurück setzen?

john30, 2024-06-15:
"flooded" seems to be bit too much for something that appears every half an hour I'd say, but anyway:
please checkout the [new version just released](https://adapter.ebusd.eu/v5/ChangeLog#20240615).
especially this log entry was most probably generated for no reason, at least from adapter perspective

MichaWilk, 2024-06-16:
Nach dem aufspielen der neuen version läuft wieder alles wie gewohnt bei mir.
DANKE

worstover, 2024-06-17:
Updating to the version solves the problem for me too and makes live much easier and more comfortable again.

Many thanks at @john30. Great job!!!

closed 2024-06-17 by kenny00111