john30/ebusd-esp32 #67: eBUS Adapter v5, Firmware Version 2024-06-15 Kommunikationsprobleme

Norb1204, 2024-06-23:
Hallo,

mit der aktuellen Firmwareversion 2024-06-15 für meinen Adapter v5 wird der Scanning Prozess nicht beendet. Die ältere Firmwareversion 2024-01-06 funktioniert und er erkennt alle Teilnehmer. Ein Neustart vom ebusd service, Spannungslos usw. bringt keinen Erfolg. Ab der Version 2024-03-17 bis Version 2024-06-15 funktioniert der Scan nicht mehr korrekt. Manchmal scannt er 1 oder 2 Slaves, meistens aber keinen (siehe Bild). 
Hat sich an den Einstellungen des Adapters was geändert bzw. muss man hier was anders konfigurieren ab Version 2024-03-17?

**Firmwareversion 2024-06-15 funktioniert nicht!**
[image]

**Firmwareversion 2024-01-06 funktioniert**
[image]

john30, 2024-07-13:
did you wait long enough?
does this happen only when using the USB connection or also via wifi e.g.?

Norb1204, 2024-07-14:
I tested several times and waited about 1/2 hour. When I install the firmware version 2024-01-06 again, it takes about 1/2 minute to detect everything.
I only use USB and no WIFI.

john30, 2024-07-15:
please post a log (as text not as image) from the start of ebusd with "--lograwdata=bytes"

Norb1204, 2024-07-17:
Here the logfile, my settings:
EBUSD_OPTS="--scanconfig=full --configpath=/etc/ebusd/ -d ens:/dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_DC:54:75:63:64:C4-if00 -p 8888 -l /var/log/ebusd.log --lograwdata=bytes --logareas=network  --httpport=8889 --htmlpath=/var/ebusd/html --accesslevel=install --updatecheck=off"

[ebusd.log](https://github.com/user-attachments/files/16269296/ebusd.log)

Norb1204, 2024-09-23:
I have tested with version 2024-08-09 and 2024-08-25, still does not work. I'm now running version 2024-01-06 again, there are no problems with this version. I will keep this version (2024-01-06) and will no longer update this.

john30, 2024-10-02:
> [ebusd.log](https://github.com/user-attachments/files/16269296/ebusd.log)

the log is useless as you limited it to network facility only

john30, 2024-10-08:
please check again with the newest version that includes the massively changed USB serial driver

Norb1204, 2024-11-08:
After installing the latest software (version 2024-10-27), nothing works at all. The ebusd service can no longer be started. Going back to the old software (version 2024-01-06) now has the same problem, the ebusd service no longer starts.
Is the adapter broken now?
[image]

I still have an old V3 adapter with this adapter everything works.

Norb1204, 2024-11-11:
After changing the udev rules to work with a symbolic link, the ebusd service works again.
Unfortunately, the current firmware of the adapter does not work either (version 2024-10-27). As described at the beginning, no result is shown when/after scanning.
[image]

john30, 2024-11-13:
still can't tell much about it without having the requested logging

Norb1204, 2024-11-23:
After buying a new adapter (v6) the same problem.
[image]

As attachment the ebusd.log
[ebusd.log](https://github.com/user-attachments/files/17880024/ebusd.log)

john30, 2024-11-24:
> After buying a new adapter (v6) the same problem.

update to the [latest version](https://github.com/john30/ebusd-esp32/blob/main/CHANGELOG.md#version-2024-10-27-reported-as-14a1b) first

Norb1204, 2024-11-24:
OK, update to version 2024-10-27, but the same, see raw log.
[image]

[ebusd.log](https://github.com/user-attachments/files/17892700/ebusd.log)

john30, 2024-11-24:
what raspi is this and which os is running?
it seems that the transfer via usb takes too long, so I suspect something weird on the os.
anyway, you might want to try with an additional "--latency=20" argument to ebusd to compensate that.
another option would be to check if using wifi instead would help. if thats not an option, it would be at least good to know if that makes a difference in the first place

Norb1204, 2024-11-25:
Pi 5, 8GB with Debian 12 (bookworm)
The changes "--latency=20" was not successful. I have tested it with different latency settings but always without success.

If I disconnect the adapter from the USB several times then stop and start the ebusd service, sometimes 2 of the 5 slaves could be scanned, sometimes all 5, but mostly none?!

Everything was fine up to firmware version 2024-01-06, but not from firmware version 2024-03-17 and later.
What has changed between firmware version 2024-01-06 and 2024-03-17, maybe you can find out something about the USB connection?
I am the only one who has this problem or does everyone else use WIFI?

Norb1204, 2024-11-28:
I make the tests on USB 3.0 Port. 
After switching to USB 2.0, it looked good at first. But after restarting the Pi, the same problem again.

Norb1204, 2024-12-01:
Again: What is the difference between version 2024-01-06 and 2024-03-17 and newer?
Version 2024-01-06 and older are all working, the others versions not working.
I have tested this with adapter V5 and V6 as well as with Pi4 and Pi5.

john30, 2024-12-01:
https://adapter.ebusd.eu/v5/ChangeLog

john30, 2024-12-14:
ok, closing

closed 2024-12-14 by john30