john30/ebusd-esp32 #69: Meshed Wifi compatibility

puni2k, 2024-07-21:
Hi there,

i'm running an AVM meshed Wifi and am experiencing some issues where my ebus Adapter Shield V5 likes to not pick the AP with the strongest signal after booting.

As far as I understand the firmware should already be able to pick the strongest AP and not the first one it finds for a given SSID. Since i operate a small army of ESPs this phenomenon isn't something new to me and i suspect an issue in the esp-idf code or an incompatibility with AVM meshes.

However i found that ESPs that have 802.11k and 802.11v enabled usually behave better and will eventually migrate to the proper AP.
I don't know though if the ESP SKU on the ebus Adapter Shield supports this.

If you would like to have a look, here's a contribution I made to a different ESP32 based project
https://github.com/meatpiHQ/wican-fw/pull/76/commits/f2e03749dfa95a67d6b29125fdda6f264bcb16b7

Sidenote: Thank you for this piece of hardware. It's the only alternative to the overpriced Wolf ISM (currently unobtanium) and was really easy to integrate.

john30, 2024-09-05:
the preference on the strongest signal is already built in. you can preselect the channel to be used.

puni2k, 2024-09-05:
Hi,
thanks for your response.

Just as an FYI: From my experience roaming support still offers some additional benefits.
From my experience meshed Wifi networks come with additional challenges and your options mentioned unfortunately are sometimes not enough.

An ESP might initially connect to the strongest AP (as mentioned, sometimes it still doesn't) but also might be forced to change to a different AP during operation. This might be caused due to an AP power cycling or something similar (Sidenote: Some APs will auto-disconnect a client if they have been connected for too long for security reasons). 
There is an edge case were the signal is still strong enough for the ESP to stay connected to a bad signal AP but the ebusd Adapter Shield webinterface becomes unresponsive and latency for the ebusd connection spikes.
With roaming support the ESP would be able to notice a better AP becoming available, without it would stay on the 'bad' AP.

Also preselecting a Wifi channel doesn't help in all cases. I.e. if you use single radio wifi repeaters.
Or if you have enough APs for a given SSID you quickly run out of non-overlapping channels anyway and have to share channel IDs between some.

But please don't feel forced to rush anything. I'm planning to switch to an Ethernet connection anyway, so this very problem won't affect me much longer. Only wanted to give some food for thought. Maybe it would help someone else.

closed 2026-01-04 by john30