john30/ebusd-esp32 #70: No eBus signal with  ESP32-C6

HandcoderWalter, 2024-09-15:
Hi,
I have installed the new C6 adapter in my Vaillant arotherm ebus network. I use an ethernet connection with W5500. I have a connection to the adapter and can configure it without problems using the web interface. However, it notifies that there is no eBUS signal. The ebus itself is actually working properly, as I have a proper connection to the myVaillant App and also to the HA integration via myVaillant. I have also measured the cabling on the soldering of the ebus plug of the board. Everything fine.
I have also tried out the workarounds mentioned for the V5 (switching to USB serial), but no success. When setting ebus -v in the REPL it prints  

init_ebus: detected: no edges

ebus: history: @-1465323 <00

The adapter seems to send some pings or similar. The blue led blinks.
HA ebusd connection seems to be fine. It works but there are of course no entities as the bus connection is not established.
 
Any ideas???

john30, 2024-10-02:
are you sure the eBUS connector has power at all? and that you connected to the right one?

HandcoderWalter, 2024-10-09:
In fact, there was something wrong with the cabling. It Workshop now properly.

closed 2024-10-09 by HandcoderWalter