john30/ebusd-esp32 #71: unable to load from NVS

M4271, 2024-09-22:
The first time I used the ebus adapter I was able to use the easi interface and connect to my wifi. 

Now, the second time (some weeks later) I am not able to connect anymore. When I use the easi interface for most of the commands I receive the following:

```
easi> sta
error: unable to load from NVS
Command returned non-zero error code: 0x1 (ERROR)
```

I did flash the latest FW  2024-08-25 but it did not fix this issue.

Anyone knows what could cause this issue? Might this be a HW defect?

john30, 2024-09-22:
seems like your wifi params were lost. the easiest way of getting them in again is by using the flash page and connect to the wifi again

john30, 2025-04-29:
closed for inactivity

closed 2025-04-29 by john30