john30/ebusd-esp32 #73: Wifi STA conection not restored after power failure

wrongisthenewright, 2024-10-22:
In this days I had a couple of power failures at home (some botched electrical work I made) and in both instances the adapter powered on but didn't reconnect to the wifi router, even after 10s of minutes.
The wifi settings were maintained but I needed to connect to the AP and reboot the adapter from the web GUI to triger a reconnect (no config changes were made).
Can it be that the adapter came back online before the router SSID came back online?
Are there any setting to force the reconnect after an initial failure?

john30, 2024-10-26:
I tried it and can't reproduce. would be helpful to get the adapter logs by using usb serial

wrongisthenewright, 2024-10-26:
Hello, 
Hope it's the right log you're asking for. HW adapter shield 1.2.4(?), FW rel 20241008.

- disconnected and powered off the adapter
- disabled SSID on the router (both 2.4 and 5ghz)
- booted adapter with power over USB
- waited some minute
- enabled SSID on the router (both bands)
- waited some minute

The adapter became available via AP feature but remained offline on STA.
After a subsequent power cycle STA connection went smoothly.

[screenlog-Failed.txt](https://github.com/user-attachments/files/17530590/screenlog-Failed.txt)

john30, 2024-10-26:
ah, wasn't aware that you configured the AP in parallel. in this case there is no retry for connecting to the station at all (only once at the beginning), because having both active at the same time will degrade the signal level a lot.
anyway, I'll remove that limit as it is intentional when someone configures both to be active

wrongisthenewright, 2024-10-26:
OK, cleared the issue. To be honest I enabled the AP in parallel with STA when I moved the adapter in a position very near my router so I didn't notice connection problems. When the first power loss occurred I was happy to not needing to open the wiring box to power cycle the adapter, being able to do it remotely.
I think that in some corner cases leaving AP and STA both active can be useful, you may prompt a warning somewhere if both feature are active at the same time.
Many thanks

john30, 2024-10-27:
solved with [2024-10-27](https://github.com/john30/ebusd-esp32/blob/main/CHANGELOG.md#version-2024-10-27-reported-as-14a1b)

closed 2024-10-27 by john30