john30/ebusd-esp32 #74: Buffer overflow

Zolli, 2024-10-22:
Hey, in an another issue i mentioned that sometimes i get buffer overflows, that somehow completly kills my system bus, only restart can solve it. On the last occasion i was able to capture soem logs that might be helpful:

Adapter: ESP32 C6
FW Version: 20241008
Connection: USB Serial

```
Ebusd  | 2024-10-22 22:31:53.187 [update notice] received update-read ignored ign6 QQ=13:
Ebusd  | 2024-10-22 22:31:53.191 [bus notice] <13fe20100c059100000691000007910000c3
Ebusd  | 2024-10-22 22:31:53.396 [update notice] received unknown MS cmd: 131e2000086e13ca46cb46d84b / 060f00000f0013
Ebusd  | 2024-10-22 22:31:53.401 [bus notice] <131e2000086e13ca46cb46d84b1800060f00000f0013ab00
Ebusd  | 2024-10-22 22:31:53.699 [update notice] received read heatpump heatpump_flow_switch QQ=13: open
Ebusd  | 2024-10-22 22:31:53.703 [bus notice] <131e2000020c455300020100b700
Ebusd  | 2024-10-22 22:31:54.105 [update notice] received unknown MS cmd: 7084200008d140cc4bc89ccc9a / 050f01080000
Ebusd  | 2024-10-22 22:31:54.109 [bus notice] <7084200008d140cc4bc89ccc9af100050f010800008b00
Ebusd  | 2024-10-22 22:31:54.180 [update notice] received update-read ignored ign5 QQ=70:
Ebusd  | 2024-10-22 22:31:54.214 [bus notice] <70fe20500b55550202000000000000003c
Ebusd  | 2024-10-22 22:31:55.068 [bus notice] <3047757275204d656469746174696f6e204572726f723a20436f726520203020...
Ebusd  | 2024-10-22 22:31:55.068 [bus notice] ...<70616e69632765642028537461636b2070726f74656374696f6e206661756c...
Ebusd  | 2024-10-22 22:31:55.068 [bus notice] ...<74292e200d0a0d0a446574656374656420696e207461736b20226c65647322...
Ebusd  | 2024-10-22 22:31:55.069 [bus error] device status: buffer overflow
Ebusd  | 2024-10-22 22:31:55.069 [bus notice] ...<20617420307834303033626538360d0a537461636b20706f696e7465723a38...
Ebusd  | 2024-10-22 22:31:55.069 [bus notice] ...<3335393730202d20307834303833363136300d0a0d0a0d0a436f7265202030...
Ebusd  | 2024-10-22 22:31:55.069 [bus notice] ...<2072656769737465722064756d703a0d0a4d455043202020203a2030783430...
Ebusd  | 2024-10-22 22:31:55.069 [bus notice] ...<303362656263202052412020202020203a2030783430303365323261202053...
Ebusd  | 2024-10-22 22:31:55.069 [bus notice] ...<502020202020203a2030783430383335373230202047502020202020203a20...
Ebusd  | 2024-10-22 22:31:55.069 [bus notice] ...<3078343038316536333420200d0a54502020202020203a2030783430383336...
Ebusd  | 2024-10-22 22:31:55.069 [bus notice] ...<313630202054302020202020203a2030783430383039366261202054312020...
Ebusd  | 2024-10-22 22:31:55.069 [bus notice] ...<202020203a2030786666666666666530202054322020202020203a20307830...
Ebusd  | 2024-10-22 22:31:55.069 [bus notice] ...<3030303030303020200d0a53302f46502020203a2030783430383261336130...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<202053312020202020203a2030783030303030303030202041302020202020...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<203a2030783430383336316463202041312020202020203a20307834303833...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<3562363820200d0a41322020202020203a2030783432306530323234202041...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<332020202020203a2030783430383336306238202041342020202020203a20...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<30783030303030303030202041352020202020203a20307832303030313039...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<3020200d0a41362020202020203a2030783030303030303063202041372020...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<202020203a2030783030303030303134202053322020202020203a20307834...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<32306530323234202053332020202020203a20307834303833356236382020...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<0d0a53342020202020203a2030783430383336316463202053352020202020...
Ebusd  | 2024-10-22 22:31:55.070 [bus notice] ...<203a2030783030303030303030202053362020202020203a20307834303833...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<39333363202053372020202020203a203078343038323930303020200d0a53...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<382020202020203a2030783030303030303030202053392020202020203a20...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<30783030303030303030202053313020202020203a20307830303030303030...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<30202053313120202020203a203078303030303030303020200d0a54332020...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<202020203a2030783430383239303030202054342020202020203a20307830...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<30303030303031202054352020202020203a20307834303832393030302020...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<54362020202020203a203078303030303030303120200d0a4d535441545553...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<203a203078303030303138383120204d545645432020203a20307834303830...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<3030303120204d434155534520203a203078303030303030316220204d5456...
Ebusd  | 2024-10-22 22:31:55.071 [bus notice] ...<414c2020203a203078336662313265323320200d0a4d484152544944203a20...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<3078303030303030303020200d0a0d0a537461636b206d656d6f72793a0d0a...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<34303833353732303a20307835363230373332642030783435353534633431...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<20307834303833303035642030783030303030303563203078303030303030...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<30302030783430383335373330203078303030303030303020307830303030...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<303030300d0a34303833353734303a20307830303030303030302030783430...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<38333537343820307866666666666666662030783430383335373438203078...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<34303833353734382030783030303030303030203078343038333537356320...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<307866666666666666660d0a34303833353736303a20307834303833353735...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<63203078343038333537356320307830303030303030312030783030303030...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<30303120307830303030303030302030783030303066666666203078303030...
Ebusd  | 2024-10-22 22:31:55.072 [bus notice] ...<303030303020307830303030303030300d0a34303833353738303a20307830...
Ebusd  | 2024-10-22 22:31:55.073 [bus notice] ...<30303030303031203078303030303030303020307834303833353732382030...
Ebusd  | 2024-10-22 22:31:55.073 [bus notice] ...<78303030303030313420307836303030373030302030783030303030303030...
Ebusd  | 2024-10-22 22:31:55.073 [bus notice] ...<203078303030303030303420307830303030303030300d0a34303833353761...
Ebusd  | 2024-10-22 22:31:55.073 [bus notice] ...<303a2030783030303030303031203078303030303030306320307834303833...
Ebusd  | 2024-10-22 22:31:55.073 [bus notice] ...<35376238203078343038333630623820307834303833353761302030783030...
Ebusd  | 2024-10-22 22:31:55.073 [bus notice] ...<303030303063203078306235363030306120307830303030303030300d0a34...
Ebusd  | 2024-10-22 22:31:55.073 [bus notice] ...<303833353763303a2030783430383336633034203078303030303030306320...
Ebusd  | 2024-10-22 22:31:55.073 [bus notice] ...<30783030303030303030203078363436353663303120307834303833303036...
Ebusd  | 2024-10-22 22:31:55.073 [bus notice] ...<33203078303030303030333820307834303832393930302030783432306432...
[... 166 lines cut ...]
Ebusd  | 2024-10-22 22:33:14.016 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:14.358 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:16.014 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:16.371 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:18.028 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:18.378 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:20.035 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:20.377 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:22.032 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:22.377 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:24.033 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:24.387 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:26.043 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:26.394 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:28.051 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:28.394 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:30.050 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:30.408 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:32.003 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:32.413 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:34.008 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:34.424 [bus notice] signal acquired
Ebusd  | 2024-10-22 22:33:37.053 [bus error] signal lost
Ebusd  | 2024-10-22 22:33:39.078 [bus error] device status: transport closed
```

john30, 2024-10-25:
please check again while turning off the RX/TX LED in the web UI

Zolli, 2024-10-25:
Thanks @john30, I'm going to try this today.

Zolli, 2024-10-26:
Hey @john30, just wanted to let you know that the adapter running smoothly more then a day now, so just wanted to say thanks for your help. Just one question, what did you think, is this fixable with firmware changes or is that bug HW related?

john30, 2024-10-26:
looks like there was an issue with simultaneous access to the RMT device which the driver is unable to handle, at least that's my best guess for now.
so in addition to preventing simultaneous access in the first place, I've also updated the strip driver as well as the IDF to latest 5.3. it is now under test for a couple of days and hopefully the measures help.
if anyone wants to do the same, then update to that version by pointing your browser at "http://adapterip/?file=testingc6.bin" (replacing "adapterip" with you know what) and then using the "Update to specific file" option in the firmware update dialog.

john30, 2024-10-27:
solved with [2024-10-27](https://github.com/john30/ebusd-esp32/blob/main/CHANGELOG.md#version-2024-10-27-reported-as-14a1b)

closed 2024-10-27 by john30