john30/ebusd-esp32 #75: Restarting the ESP32-C6 adapter and causing the external probe to drop and the machine to stop

edudele, 2024-10-24:
Something is happening in the latest firmware versions. I have adapter reboots and the air source heat pump stops and the external probe drops to -40º.The bus does not seem to go down but the adapter restart affects the operation of the machine.

**If you need more information than I provide, ask me.**

daemon mounted on docker

It has the same behavior both by USB and WIFI and I think that V5 is also affected.

Adapter:
Chip ID: 543204291658, ESP32-C6, rev 1
Hostname: ebusd-c6
Up time: 30
Free heap: 192044 / 367608
ebusd connected: yes
eBUS signal: acquired
MQTT connected: yes

ADAPTER LOG (It seems to be a normal start)

12:20:08 |   | [5n
-- | -- | --
12:20:08 | Info | replExecuteFromFile: executing commands from '/data/boot.eas'
12:20:08 | Info | execute: strip -m 1 -b 255
12:20:08 | Info | gpio: GPIO[23]\| InputEn: 0\| OutputEn: 1\| OpenDrain: 0\| Pullup: 1\| Pulldown: 0\| Intr:0
12:20:08 | Info | execute: function ledb "@strip -b"
12:20:08 |   | function 'ledb' installed
12:20:08 | Info | execute: function ledg "@strip -g"
12:20:08 |   | function 'ledg' installed
12:20:08 | Info | execute: led -p 16 -i 0 255
12:20:08 |   | 0 duty set to 254
12:20:08 | Info | execute: function ledboot "@led 0"
12:20:08 |   | function 'ledboot' installed
12:20:08 | Info | execute: function leds "@strip -b" "@led 0"
12:20:08 |   | function 'leds' installed
12:20:08 | Info | execute: io -iue 9
12:20:08 |   | IO 9 direction set to 01
12:20:08 |   | IO 9 pull set to 00
12:20:08 |   | IO 9 = 1
12:20:08 | Info | execute: io -iu 6
12:20:08 |   | IO 6 direction set to 01
12:20:08 |   | IO 6 pull set to 00
12:20:08 |   | IO 6 = 1
12:20:08 | Info | execute: nvs_namespace easi
12:20:08 | Info | cmd_nvs: Namespace set to 'easi'
12:20:08 | Info | replExecuteFromFile: successfully executed 9/9 commands from '/data/boot.eas'
12:20:08 | Info | replExecuteFromFile: executing commands from '/data/main.eas'
12:20:08 | Info | execute: var skipstart 0
12:20:08 | Info | execute: leds 255 0 700
12:20:08 | Info | execute: strip -b 255 0 700
12:20:08 | Info | execute: led 0 255 0 700
12:20:08 |   | 0 duty set to 254
12:20:08 | Info | execute: sleep 1000
12:20:08 | Info | main_task: Returned from app_main()
12:20:09 | Info | execute: if "isio -l 9" "var skipstart 1"
12:20:09 | Info | execute: isio -l 9
12:20:09 |   | IO 9 level 1 is unmet
12:20:09 | Info | execute: leds 255 0 700
12:20:09 | Info | execute: strip -b 255 0 700
12:20:09 | Info | execute: led 0 255 0 700
12:20:09 |   | 0 duty set to 254
12:20:09 | Info | execute: sleep 1000
12:20:10 | Info | execute: if "isio -l 9" "var skipstart 1"
12:20:10 | Info | execute: isio -l 9
12:20:10 |   | IO 9 level 1 is unmet
12:20:10 | Info | execute: leds 255 0 700
12:20:10 | Info | execute: strip -b 255 0 700
12:20:10 | Info | execute: led 0 255 0 700
12:20:10 |   | 0 duty set to 254
12:20:10 | Info | execute: sleep 1000
12:20:11 | Info | execute: if "isio -l 9" "var skipstart 1"
[... 300 lines cut ...]
{"log":"2024-10-24 12:20:56.418 [update notice] received unknown MS cmd: 7108b51a03046334 / 0e63cf2000c0022101001800061800\n","stream":"stdout","time":"2024-10-24T10:20:56.41868698Z"}
{"log":"2024-10-24 12:20:56.551 [update notice] received unknown MS cmd: 1076b516081002ffff03044131 / 0b0200010304413100000000\n","stream":"stdout","time":"2024-10-24T10:20:56.552104816Z"}
{"log":"2024-10-24 12:20:56.804 [update notice] received unknown MS cmd: 7108b51a03046435 / 0e6400000000000100900300000000\n","stream":"stdout","time":"2024-10-24T10:20:56.804686792Z"}
{"log":"2024-10-24 12:20:56.936 [update notice] received unknown MS cmd: 1076b516081002ffff03042131 / 0b0201010304213100000000\n","stream":"stdout","time":"2024-10-24T10:20:56.936756455Z"}
{"log":"2024-10-24 12:20:57.189 [update notice] received unknown MS cmd: 7108b51a03046536 / 0e6500000000000000000000000000\n","stream":"stdout","time":"2024-10-24T10:20:57.189269025Z"}
{"log":"2024-10-24 12:20:57.322 [update notice] received unknown MS cmd: 1076b516081003ffff03042130 / 0b0300010304213000000000\n","stream":"stdout","time":"2024-10-24T10:20:57.322395194Z"}
{"log":"2024-10-24 12:20:57.630 [update notice] received unknown MS cmd: 1076b516081003ffff0304212e / 0b0301010304212e00000000\n","stream":"stdout","time":"2024-10-24T10:20:57.630700681Z"}
{"log":"2024-10-24 12:20:57.725 [update notice] received read hmu00 State QQ=71: 0;123;193;8\n","stream":"stdout","time":"2024-10-24T10:20:57.72600665Z"}
{"log":"2024-10-24 12:20:57.989 [update notice] received unknown MS cmd: 1076b516081000ffff03042130 / 0b0000010304123100000000\n","stream":"stdout","time":"2024-10-24T10:20:57.989723005Z"}
{"log":"2024-10-24 12:20:58.254 [update notice] received read hmu00 Status01 QQ=10: 31.5;32.0;-;-;-;off\n","stream":"stdout","time":"2024-10-24T10:20:58.254560274Z"}
{"log":"2024-10-24 12:20:58.523 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff0080ff400000ff\n","stream":"stdout","time":"2024-10-24T10:20:58.52348504Z"}
{"log":"2024-10-24 12:20:58.790 [update notice] received unknown MS cmd: 1076b512030f0001 / 070002000080ff03\n","stream":"stdout","time":"2024-10-24T10:20:58.790513158Z"}
{"log":"2024-10-24 12:20:59.098 [update notice] received unknown MS cmd: 1076b516081002ffff03054131 / 0b1200ff0305413100000000\n","stream":"stdout","time":"2024-10-24T10:20:59.098549691Z"}
{"log":"2024-10-24 12:20:59.362 [update notice] received update-write hmu00 SetMode QQ=10: auto;0.0;-;-;1;1;1;0;0;0\n","stream":"stdout","time":"2024-10-24T10:20:59.362813651Z"}
{"log":"2024-10-24 12:20:59.627 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101\n","stream":"stdout","time":"2024-10-24T10:20:59.628091139Z"}
{"log":"2024-10-24 12:20:59.955 [update notice] received unknown MS cmd: 1076b516081002ffff03052131 / 0b1201ff0305213100000000\n","stream":"stdout","time":"2024-10-24T10:20:59.955349615Z"}
{"log":"2024-10-24 12:21:00.241 [update notice] received unknown MS cmd: 1076b516081003ffff03052130 / 0b1300ff0305213000000000\n","stream":"stdout","time":"2024-10-24T10:21:00.241848817Z"}
{"log":"2024-10-24 12:21:00.548 [update notice] received unknown MS cmd: 1076b516081003ffff0305212e / 0b1301ff0305212e00000000\n","stream":"stdout","time":"2024-10-24T10:21:00.549246551Z"}
{"log":"2024-10-24 12:21:00.855 [update notice] received unknown MS cmd: 1076b516081000ffff03052130 / 0b1000ff0305123100000000\n","stream":"stdout","time":"2024-10-24T10:21:00.855955888Z"}
{"log":"2024-10-24 12:21:01.083 [update notice] received read hmu00 State QQ=71: 0;123;193;8\n","stream":"stdout","time":"2024-10-24T10:21:01.083516298Z"}
```
Uptime sensor Home Assistant of adapter topic **ebusd/adaptador** 

[image]
[image]

john30, 2024-10-25:
seems to be related to #74

edudele, 2024-10-25:
I have now had to change to version 3.1 of the adapter as winter has already arrived and I cannot have continuous power cuts in heating and ACS.

How can I get a log before the restart, some syslog or rsyslog?

This is a fairly serious problem for those who do not have an older adapter.
Has it been identified from which firmware version it occurs?

john30, 2024-10-26:
see the other issue and follow https://github.com/john30/ebusd-esp32/issues/74#issuecomment-2436912626

closing as duplicate

closed 2024-10-26 by john30