john30/ebusd-esp32 #77: After a power outage the adapter does not work

edudele, 2024-10-28:
Good morning, After a power outage the adapter does not work

Version:

Chip ID: 543204291658, [ESP32-C6, rev 1](http://192.168.1.202/#/status?verbose)
Hostname: ebusd-c6
Up time: 886
Free heap: 190288 / 366744
ebusd connected: yes (inactive)
eBUS signal: acquired
MQTT connected: yes

Build: [20241027](https://adapter.ebusd.eu/v5-c6/ChangeLog#20241027)
[ebusd device string](https://github.com/john30/ebusd/wiki/2.-Run#device-options): "ens:/dev/ttyACM0" (number may vary)
WiFi station/client: VILLAMANIN, 192.168.1.202, 96% (-52dBm)
WiFi access point: inactive

`

11:25:23 |   | [5n
-- | -- | --
11:25:23 | Info | replExecuteFromFile: executing commands from '/data/boot.eas'
11:25:23 | Info | execute: strip -m 1 -b 255
11:25:23 | Info | gpio: GPIO[23]\| InputEn: 0\| OutputEn: 1\| OpenDrain: 0\| Pullup: 1\| Pulldown: 0\| Intr:0
11:25:23 | Info | execute: function ledb "@strip -b"
11:25:23 |   | function 'ledb' installed
11:25:23 | Info | execute: function ledg "@strip -g"
11:25:23 |   | function 'ledg' installed
11:25:23 | Info | execute: led -p 16 -i 0 255
11:25:23 |   | 0 duty set to 254
11:25:23 | Info | execute: function ledboot "@led 0"
11:25:23 |   | function 'ledboot' installed
11:25:23 | Info | execute: function leds "@strip -b" "@led 0"
11:25:23 |   | function 'leds' installed
11:25:23 | Info | execute: io -iue 9
11:25:23 |   | IO 9 direction set to 01
11:25:23 |   | IO 9 pull set to 00
11:25:23 |   | IO 9 = 1
11:25:23 | Info | execute: io -iu 6
11:25:23 |   | IO 6 direction set to 01
11:25:23 |   | IO 6 pull set to 00
11:25:23 |   | IO 6 = 1
11:25:23 | Info | execute: nvs_namespace easi
11:25:23 | Info | cmd_nvs: Namespace set to 'easi'
11:25:23 | Info | replExecuteFromFile: successfully executed 9/9 commands from '/data/boot.eas'
11:25:23 | Info | replExecuteFromFile: executing commands from '/data/main.eas'
11:25:23 | Info | execute: var skipstart 0
11:25:23 | Info | execute: leds 255 0 700
11:25:23 | Info | execute: strip -b 255 0 700
11:25:23 | Info | execute: led 0 255 0 700
11:25:23 |   | 0 duty set to 254
11:25:23 | Info | execute: sleep 1000
11:25:23 | Info | main_task: Returned from app_main()
11:25:24 | Info | execute: if "isio -l 9" "var skipstart 1"
11:25:24 | Info | execute: isio -l 9
11:25:24 |   | IO 9 level 1 is unmet
11:25:24 | Info | execute: leds 255 0 700
11:25:24 | Info | execute: strip -b 255 0 700
11:25:24 | Info | execute: led 0 255 0 700
11:25:24 |   | 0 duty set to 254
11:25:24 | Info | execute: sleep 1000
11:25:25 | Info | execute: if "isio -l 9" "var skipstart 1"
11:25:25 | Info | execute: isio -l 9
11:25:25 |   | IO 9 level 1 is unmet
11:25:25 | Info | execute: leds 255 0 700
11:25:25 | Info | execute: strip -b 255 0 700
11:25:25 | Info | execute: led 0 255 0 700
11:25:25 |   | 0 duty set to 254
11:25:25 | Info | execute: sleep 1000
11:25:26 | Info | execute: if "isio -l 9" "var skipstart 1"
[... 155 lines cut ...]
> 2024-10-28 11:43:22.902 [bus error] send to fe: ERR: read timeout, retry
> 2024-10-28 11:43:22.924 [bus error] send to fe: ERR: wrong symbol received
> 2024-10-28 11:43:22.924 [main error] initial scan failed: ERR: wrong symbol received
> 2024-10-28 11:43:23.608 [update notice] received unknown MS cmd: 7108b5110107 / 0a00b60081003600000000
> 2024-10-28 11:43:23.822 [update notice] received unknown MS cmd: 0376b512061300124a0300 / 0200ff
> 2024-10-28 11:43:24.306 [update notice] received unknown MS cmd: f108b5110101 / 093b3c0080ffff0000ff
> 2024-10-28 11:43:24.483 [update notice] received unknown MS cmd: f108b5110100 / 09df0112000808000000
> 2024-10-28 11:43:25.712 [update notice] received unknown MS cmd: 1008b5110101 / 093b3c0080ffff0000ff
> 2024-10-28 11:43:25.981 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff0080ff410000ff
> 2024-10-28 11:43:26.248 [update notice] received unknown MS cmd: 1076b512030f0101 / 070902000080ff03
> 2024-10-28 11:43:26.387 [update notice] received unknown MS cmd: 7108b5110107 / 0a00b60081003600000000
> 2024-10-28 11:43:26.564 [update notice] received unknown MS cmd: 1008b51009000045ffffff060000 / 0101
> 2024-10-28 11:43:26.829 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
> 2024-10-28 11:43:29.688 [update notice] received unknown MS cmd: 7108b5110107 / 0a00b60081003600000000
> 2024-10-28 11:43:30.313 [update notice] received unknown MS cmd: f115b52406020000004800 / 06010048000100
> 2024-10-28 11:43:30.499 [update notice] received unknown MS cmd: f115b52406020002001e00 / 0601021e000100
> 2024-10-28 11:43:30.624 [update notice] received unknown MS cmd: 0376b512061300124a0300 / 0200ff
> 2024-10-28 11:43:30.730 [update notice] received unknown MS cmd: f115b52406020003002200 / 08030322000000b041
> 2024-10-28 11:43:30.924 [update notice] received unknown MS cmd: f115b52406020003001400 / 08010314000000b041
> 2024-10-28 11:43:31.110 [update notice] received unknown MS cmd: f115b52406020003000600 / 06030306000200
> 2024-10-28 11:43:31.296 [update notice] received unknown MS cmd: f115b52406020003001b00 / 0601031b000000
> 2024-10-28 11:43:31.481 [update notice] received unknown MS cmd: f115b52406020001000f00 / 0601010f000000
> 2024-10-28 11:43:32.924 [main notice] starting initial broadcast scan
> 2024-10-28 11:43:33.015 [bus error] send to fe: ERR: read timeout, retry
> 2024-10-28 11:43:33.628 [bus error] send to fe: ERR: read timeout, retry

edudele, 2024-10-28:
It would be possible to generate a topic within the adapter to send to force a restart of the adapter itself.
[image]

john30, 2024-10-30:
how is ebusd started, i.e. with which arguments?

edudele, 2024-10-30:
After the power outage, possibly the daemon was not even started, it is with docker on a minipc, but the adapter

john30, 2024-11-13:
that does not answer my question, but anyway:
maybe you did not configure the container to restart after reboot

edudele, 2024-11-13:
> that does not answer my question, but anyway: maybe you did not configure the container to restart after reboot

When we get to opening an issues it's because I have a real problem and I want to help make this work better. Here is what you asked for, I have been studying ebusd and its files for 2-3 years, to send us with that answer.

As I said before, it is not the container.
Here is my docker compose:

```
version: '3' 
services:
    ebusd:
        container_name: ebusdv5
        restart: unless-stopped
        #image: john30/ebusd:devel
        image: john30/ebusd:latest
        #image: john30/ebusd:v23.2
        privileged: true
        environment:
          - EBUSD_SCANCONFIG=full
          - EBUSD_CONFIGPATH=/ebusd/ebusd-configuration/ebusd-2.1.x/en
          #- EBUSD_LOGAREAS=all
          #- EBUSD_LOGLEVEL=info
          - EBUSD_ACCESSLEVEL=*
          - EBUSD_DEVICE=ens:/dev/ttyACM0
          #- EBUSD_DEVICE=ens:192.168.1.202:9999
          - EBUSD_MQTTHOST=mosquitto
          - EBUSD_MQTTPORT=1883
          - EBUSD_MQTTUSER=xxxxx
          - EBUSD_MQTTPASS=xxxx
          - EBUSD_MQTTJSON=
          - EBUSD_LATENCY=10
          - EBUSD_MQTTINT=/ebusd/ebusd-configuration/mqtt-hassio.cfg
          #- EBUSD_MQTTTOPIC=
        volumes:
          - /docker/ebusd/:/ebusd
          - /etc/localtime:/etc/localtime:ro
        devices:
          - /dev/ttyACM0:/dev/ttyACM0
```

john30, 2024-11-15:
> I have been studying ebusd and its files for 2-3 years, to send us with that answer.

no idea what you're trying to tell me with this.

> As I said before, it is not the container.

which contradicts with what you wrote before:
> ...possibly the daemon was not even started yet...

and still you didn't provide the logging I asked for, so how would I know whats going on??
and did you check after the reboot that the container was actually running?

edudele, 2024-11-16:
luck

closed 2024-11-16 by edudele