john30/ebusd-esp32 #78: Unable to get readings after the update to 24.1

VivantSenior, 2024-10-28:
Hi,

I was unable to OTA flash [eBUS Adapter Shield v5](https://adapter.ebusd.eu/v5/index.en.html) with 24.1 because of an error:
> ebusd Status: begin failed, error: ESP_ERR_HTTP_CONNECT

After a few tries I have done upgrade via the USB cable and now can't read any data from the adapter:

Build: [20241027](https://adapter.ebusd.eu/v5/ChangeLog#20241027)
[ebusd device string](https://github.com/john30/ebusd/wiki/2.-Run#device-options): "ens:192.168.50.201:9999" or "mdns:dc56012169e1" since ebusd 24.1
WiFi station/client: wifi_ssid, 192.168.50.201, 100% (-34dBm)
WiFi access point: inactive

Chip ID: dc56012169e1, ESP32-C3, rev 4
Hostname: ebus-host
PCB: 1.24.1
Build: 20241027
Flash: 4194304
IDF: v5.3.1-638-ga0f798cfc4-dirty
Reset reason: 5
eBUS r/w: 5577/15
Host r/w: 148/9624
Up time: 208
Free heap: 35392 / 255652
ebusd connected: yes (inactive)
eBUS signal: acquired

|Time | Icon | Log |
| -------- | ------- | ------- |
00:17:30 |   | 2 duty set to 254
00:17:30 | Info | execute: sleep 1000
00:17:30 | Info | leds: duty set to 254
00:17:31 | Info | execute: if "isio -l 9" "var skipstart 1"
00:17:31 | Info | execute: isio -l 9
00:17:31 |   | IO 9 level 1 is unmet
00:17:31 | Info | execute: var rpi 0
00:17:31 | Info | execute: if "isio -l 6" "var rpi 1"
00:17:31 | Info | execute: isio -l 6
00:17:31 |   | IO 6 level 1 is unmet
00:17:31 | Info | execute: var isnew 1
00:17:31 | Info | execute: if "§isnvs ebus.host" "var isnew 0"
00:17:31 | Info | execute: isnvs ebus.host
00:17:31 | Info | execute: var isnew 0
00:17:31 |   | variable isnew = 0
00:17:31 | Info | execute: if "§isnvs sta.on" "var isnew 0"
00:17:31 | Info | execute: isnvs sta.on
00:17:31 | Info | execute: var isnew 0
00:17:31 | Info | execute: if "§isnvs eth.on" "var isnew 0"
00:17:31 | Info | execute: isnvs eth.on
00:17:31 | Warning | execute: command failed: isnvs eth.on = 1
00:17:31 | Info | execute: on -lo iol 9 "leds 255" "sleep 500" restart
00:17:31 |   | io trigger installed
00:17:31 | Info | execute: on iol 9 "ledboot 50 0"
00:17:31 |   | io trigger installed
00:17:31 | Info | execute: function start "ledg 50 0 3000" server mqtt
00:17:31 |   | function 'start' installed
00:17:31 | Info | execute: on -o wifi 1 start
00:17:31 |   | network trigger installed
00:17:31 | Info | execute: on -o eth 1 start
00:17:31 |   | network trigger installed
00:17:31 | Info | execute: on wifi 0 "ledb 0"
00:17:31 |   | network trigger installed
00:17:31 | Info | execute: on eth 0 "ledb 0"
00:17:31 |   | network trigger installed
00:17:31 | Info | execute: on -o server 2 "after -o -m 500 0 restart"
00:17:31 |   | server trigger installed
00:17:31 | Info | execute: if "var skipstart" "ledb 255 0 5000" "ledboot 50" "ap EBUS" start exit
00:17:31 | Info | execute: var skipstart
00:17:31 |   | variable skipstart = 0
00:17:31 | Info | execute: if "isnvs sta.on" "ledb 10 255 5000" "sleep 100" sta
00:17:31 | Info | execute: isnvs sta.on
00:17:31 | Info | execute: ledb 10 255 5000
00:17:31 | Info | execute: led 0 10 255 5000
00:17:31 |   | 0 duty requested to 10
00:17:31 | Info | execute: sleep 100
00:17:31 | Info | leds: duty set to 10
00:17:31 | Info | execute: sta
00:17:31 | Warning | event: handler already registered, overwriting
00:17:31 |   | using individual IP: 192.168.50.201 / 255.255.255.0, gateway 192.168.50.1
00:17:31 |   | sta connecting to 'wifi_ssid'
00:17:31 | Warning | wifi:
00:17:31 |   | Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
00:17:35 | Info | cmd_wifi: sta connect, set fix ip
00:17:35 | Info | repl_event_task: sta event: 1
00:17:35 | Info | execute: start
00:17:35 | Info | execute: ledg 50 0 3000
00:17:35 | Info | execute: led 1 50 0 3000
00:17:35 |   | 1 duty set to 50
00:17:35 | Info | execute: server
00:17:35 | Info | execute: if "var isnew" "ap EBUS" "sta -w" else "if \"isnvs ap.on\" ap"
00:17:35 | Info | execute: var isnew
00:17:35 |   | variable isnew = 0
00:17:35 | Info | execute: if "isnvs ap.on" ap
00:17:35 | Info | execute: isnvs ap.on
00:17:35 | Info | execute: var eth 0
00:17:35 | Info | execute: if "eth -t" "var eth 1" else "if \"eth -wt\" \"var eth 2\""
00:17:35 | Info | execute: eth -t
00:17:35 | Info | initialise_mdns: mdns configured
00:17:35 | Info | esp-rest: Starting HTTP Server
00:17:35 |   | server started
00:17:35 | Info | execute: mqtt
00:17:35 | Info | sntpTimeCb: got NTP time: 2024-10-28 23:17:36
00:17:36 |   | Ethernet on USR-ES1 not available
00:17:36 | Warning | execute: command failed: eth -t = 1
00:17:36 | Info | execute: if "eth -wt" "var eth 2"
00:17:36 | Info | execute: eth -wt
00:17:36 | Warning | httpd_txrx: httpd_sock_err: error in send : 104
00:17:37 |   | Ethernet on W5500 module not available
00:17:37 | Warning | execute: command failed: eth -wt = 1
00:17:37 | Info | execute: io -a
00:17:37 | Info | execute: function chkethcfg "if \"eth -c\" \"var eth 1\""
00:17:37 |   | function 'chkethcfg' installed
00:17:37 | Info | execute: if "!var isnew" "if \"!var eth\" chkethcfg"
00:17:37 | Info | execute: var isnew
00:17:37 |   | variable isnew = 0
00:17:37 | Info | execute: if "!var eth" chkethcfg
00:17:37 | Info | execute: var eth
00:17:37 |   | variable eth = 0
00:17:37 | Info | execute: chkethcfg
00:17:37 | Info | execute: if "eth -c" "var eth 1"
00:17:37 | Info | execute: eth -c
00:17:37 | Warning | execute: command failed: eth -c = 1
00:17:37 | Info | execute: function enet eth "leds 50"
00:17:37 |   | function 'enet' installed
00:17:37 | Info | execute: if "isnvs eth.on" "if \"var eth\" enet"
00:17:37 | Info | execute: isnvs eth.on
00:17:37 | Warning | execute: command failed: isnvs eth.on = 1
00:17:37 | Info | execute: if "isnvs ebus.host" ebus exit
00:17:37 | Info | execute: isnvs ebus.host
00:17:37 | Info | execute: ebus
00:17:37 | Info | init_ebus: switching to enhanced eBUS mode on TCP port (with bitrate deviation 0 and enhanced arbitration delay 200 us)
00:17:37 | Info | setup_ebus_host_tcp: listening on port 9999
00:17:37 | Info | setup_uart: eBUS UART 1 on TX=10(inv) RX=7(inv) with 2400 Bd
00:17:37 | Info | uart: queue free spaces: 16
00:17:37 | Info | addMdnsService: mDNS service added
00:17:37 | Info | execute: exit
00:17:37 | Info | execute: command exit
00:17:37 | Info | execute: command exit
00:17:37 | Info | replExecuteFromFile: successfully executed 35/35 commands from '/data/main.eas'
00:17:37 |   | Welcome to eBUS Adapter Shield Interface!
00:17:38 | Info | wait_host_tcp: accepted: 192.168.50.247
00:17:38 | Info | loop: host reset
00:18:35 | Error | wifi:
00:18:35 |   | CCMP replay detected: A1=dc:54:75:61:69:d0 A2=04:42:1a:a5:79:80 PN=879, RSC=880 seq=6352
00:18:35 | Error | wifi:
00:18:35 |   | CCMP replay detected: A1=dc:54:75:61:69:d0 A2=04:42:1a:a5:79:80 PN=878, RSC=880 seq=6368
00:18:45 | Warning | read_host_tcp: closed
00:18:45 | Info | close_client: socket closed
00:18:45 | Info | loop: host disconnected
00:18:54 | Info | wait_host_tcp: accepted: 192.168.50.247
00:18:54 | Info | loop: host reset
00:19:36 | Error | wifi:
00:19:36 |   | CCMP replay detected: A1=dc:54:75:61:69:d0 A2=04:42:1a:a5:79:80 PN=1847, RSC=1848 seq=20160
00:20:41 | Warning | read_host_tcp: closed
00:20:41 | Info | close_client: socket closed
00:20:41 | Info | loop: host disconnected

Here are some logs from [Home Assistant addon](https://github.com/LukasGrebe/ha-addons):
> 2024-10-29 00:23:23.124 [bus notice] device status: extra info: firmware 1.1[4a1b].1[4a1b], jumpers 0x0b
> 2024-10-29 00:23:23.345 [mqtt notice] connection established
> 2024-10-29 00:23:23.480 [bus debug] ERR: read timeout during receive command, switching to skip
> 2024-10-29 00:23:24.678 [network info] [00001] connection closed
> 2024-10-29 00:23:24.679 [network info] [00001] client connection opened 127.0.0.1
> 2024-10-29 00:23:25.419 [bus debug] ERR: read timeout during receive command, switching to skip
> 2024-10-29 00:23:25.681 [network debug] dead connection removed - 0
> 2024-10-29 00:23:29.506 [bus debug] ERR: SYN received during receive command, switching to ready
> 2024-10-29 00:23:32.864 [main debug] performing regular tasks
> 2024-10-29 00:23:35.374 [bus debug] ERR: SYN received during receive command, switching to ready
> 2024-10-29 00:23:42.864 [main debug] performing regular tasks
> 2024-10-29 00:23:43.759 [bus debug] ERR: read timeout during receive command, switching to skip
> 2024-10-29 00:23:49.878 [bus debug] ERR: read timeout during receive command, switching to skip
> 2024-10-29 00:23:52.865 [main debug] performing regular tasks
> 2024-10-29 00:23:53.451 [bus debug] ERR: read timeout during receive command, switching to skip

john30, 2024-12-15:
please check again with the new firmware just released as there were many fixes on esp-idf wrt wifi

john30, 2025-04-29:
closed for inactivity

closed 2025-04-29 by john30

VivantSenior, 2025-04-30:
Hi, sorry for the delay. Yes, after last few updates it works as expected :)