john30/ebusd-esp32 #81: Helioset saunier Duval problem with thé signal

didjerislade, 2024-11-16:
Hi,

I would like to monitor some parameters of my Helioset but when I connect my ebus adapter v6 to the ebus port I don't get any signal. ebus connected: yes but inactive 

Has anyone had any experience with this kind of devices?

Thanks in advance.

john30, 2024-11-24:
probably insufficient support for this heating, so an [ebusd-configuration](https://github.com/john30/ebusd-configuration) issue instead.

closed 2024-11-24 by john30

adrianoapmartins, 2025-04-20:
Hey @didjerislade, did you have any luck with this?

didjerislade, 2025-04-21:
it's seems that it can't work anyway..

john30, 2025-04-24:
if the bus does not have a signal at all, you can try [--generatesyn](https://github.com/john30/ebusd/wiki/2.-Run#ebus-options)