john30/ebusd-esp32 #86: EBUS access point stays up during normal operation

jleeuwes, 2024-12-11:
I've connected a new adapter v5 to the GPIO pins of a Raspberry Pi 4 [like this](https://adapter.ebusd.eu/v5/index.en.html#rpi). I did connect to the EBUS access point to look at the easi> webinterface on <http://192.168.4.1>, however I didn't need to configure anything there for this kind of connection. I got it to work perfectly with the ebusd sofware running on the Raspberry.

However, I noticed that the passwordless EBUS access point remained available, although ebusd was running and the blue led was blinking every five seconds like explained on <https://adapter.ebusd.eu/v5/index.en.html#leds>. When I connected to it and opened <http://192.168.4.1> in a browser, a blank screen was presented and an attempt was made to download <http://192.168.4.1/assets/index-D6lQIIob.js>, which was very slow and failed. When downloading that file using curl, I got a bunch of binary data, not the javascript that is normally returned when using the easi> webinterface. I can only assume this was 'raw' ebus data?

Since don't want any passerby to be able to connect to and control my heating system, I wanted to disable this EBUS access point. I did not find any way to do this, but I figured I could configure the adapter to connect to a bogus Wi-Fi network, assuming it was not capable of running both a Wi-Fi client and a Wi-Fi access point at the same time.

This worked. The EBUS access point was gone. Strangely enough, when I switched the WIFI station/client option back off in the easi> webinterface and rebooted, the EBUS access point did not reappear.

Long story short: the EBUS access point seems to linger until _something_ is configured in the easi> webinterface once. Then the configuration can be safely reverted.

I'm happy to give any extra information you might need, or try out things, although I'm not sure how to reproduce the problem.

john30, 2024-12-14:
as long as the device is treated as "new" (i.e. no adjustment at all was made in the web UI yet), the AP stays on. in this state it tries to detecte automatically whether it sits on RPI.
a one-time save+reset action in the web ui disables the AP. see the documentation where it says "...the host connection needs to be set to Raspberry Pi..."

closed 2024-12-14 by john30

jleeuwes, 2025-01-08:
Sorry for the delay.
I found out I can 'factory reset' the device by updating to the latest firmware with the 'erase' option enabled. So I was able to try out some different things.

If the device is "new", the AP stays on and ebusd can communicate with the device. I get this in my ebusd log:

```
2025-01-08 20:56:41.484 [device notice] transport opened
2025-01-08 20:56:41.506 [main notice] ebusd 24.1.24.1 started with broadcast scan on device: /dev/ttyAMA0, ser
ial high speed, enhanced
2025-01-08 20:56:41.952 [bus notice] bus started with own address 31/36
2025-01-08 20:56:41.952 [device error] missing enhanced byte 2
2025-01-08 20:56:41.952 [device error] unexpected enhanced byte 2
2025-01-08 20:56:41.952 [bus notice] signal acquired
2025-01-08 20:56:41.955 [mqtt notice] connection established
2025-01-08 20:56:43.853 [bus notice] new master 71, master count 2
2025-01-08 20:56:43.912 [bus notice] new master 03, master count 3
2025-01-08 20:56:43.912 [update notice] received unknown MS cmd: 7108b5110107 / 0a008802c0003000000000
2025-01-08 20:56:45.926 [update notice] received unknown MS cmd: 0376b51206130010000005 / 0200ff
...
```

With the bus cable detached, this looks a little different, but still OK:

```
2025-01-08 20:56:03.998 [device notice] transport opened
2025-01-08 20:56:04.008 [main notice] ebusd 24.1.24.1 started with broadcast scan on device: /dev/ttyAMA0, ser
ial high speed, enhanced
2025-01-08 20:56:04.381 [bus notice] bus started with own address 31/36
2025-01-08 20:56:04.381 [device notice] reset, supports info
2025-01-08 20:56:04.383 [device notice] extra info: firmware 1.1[4c15].1[4c15], jumpers 0x03
```

Then I do a one-time save+reset action in the web ui. Afterwards, the EBUS AP is gone, as expected, but also there is no more communication between ebusd and the device and/or bus:

```
2025-01-08 21:02:56.062 [device notice] transport opened
2025-01-08 21:02:56.072 [main notice] ebusd 24.1.24.1 started with broadcast scan on device: /dev/ttyAMA0, serial high speed, enhanced
2025-01-08 21:02:56.482 [bus notice] bus started with own address 31/36
2025-01-08 21:02:56.499 [mqtt notice] connection established
2025-01-08 21:05:06.684 [main notice] update check: OK
```

@john30 you mention that the device automatically detects whether it sits on a Raspberry Pi _as long as the device is treated as "new"_. So maybe it no longer auto-detects this in the "normal" state (after a save+reset)? That is consistent with the behavior I am seeing, as described above. How do I set the host connection to Raspberry Pi explicitly? I cannot find anything about this in the documentation.

jleeuwes, 2025-01-11:
> By using the [easi> interface](https://adapter.ebusd.eu/v5/easi.en), the host connection needs to be set to Raspberry Pi.

I'd like to do this, but how? None of the options shown relate to the GPIO / Raspberry PI connection. [image]

john30, 2025-01-14:
not too difficult: click on the "eBUS" section

jleeuwes, 2025-02-05:
> not too difficult: click on the "eBUS" section

🤦 Thanks, I totally missed that.