john30/ebusd-esp32 #89: After some uptime some values no longer updated

t-master-t, 2024-12-20:
After some days of uptime some values are no longer read/updated.

For example it hits always:
ctlv2/hwcstoragetemp  and ctlv2/Hc1FlowTemp

Other values are still being read and provided to home assistant

Rebooting the adapter shield ie. via the webinterface of the adapter fixes the issues.

I am using Home Assistant with the ebusd and Mosquitto broker addons. 
The adapter is plugged-in via USB and uses ens:/dev/ttyACM0 as device id.

Is there any way to regularly schedule a reboot (ie. nightly) or trigger via API? This would not fix the issue but would be a fair workaround for me for the time being.

john30, 2025-01-26:
you can simply issue the same request that is executed when pressing the reboot button in the ui

t-master-t, 2025-01-27:
Hi John, this is something i thought would be more complicated as i 
could not directly see which command is being executed. But thinking 
about it i was able to figure it out with the network analysis and could 
create a curl command.

Links:
------
[1] 
https://github.com/john30/ebusd-esp32/issues/89#issuecomment-2614552411
[2] 
https://github.com/

john30, 2025-04-29:
is this still happening?

t-master-t, 2025-05-17:
Hi John, since I reboot the shield every night, it works. I cannot tell you if the newer firmwares are also fixing it without the need of a reboot. If you need me testing this, then pls. tell me. Thanks!

john30, 2025-05-17:
maybe you want to checkout the new release just posted